magicbug/Cloudlog #168: Date is off?

shadowmite, 2015-01-18:
Perhaps I'm missing something but I just installed this logger and set it up. When I go to create a QSO the date is off (appears to be adding 7 hours to my timezone). I am in a GMT-7 timezone, so I assumed this would be a controllable option. 

I went and set the config.php setting to both local and gmt. Both show the EXACT same date and time. I looked up the routine and I'm not certain what the problem is?

shadowmite, 2015-01-18:
Furthermore it appears regardless if you set gmt or local you hard fix to the UTC timezone, this is counterintuitive and needs to be stated or make the option work. 

Also, you have a setting in the config.php for a date string format, but regardless what you set the script uses d-m-Y. Again, it should use what the config.php file has been set to use.

magicbug, 2015-01-18:
Indeed UTC is hard coded for time stored in the database this is because UTC is generally the "standard" for amateur radio.

The profile time zone selection is so that the live times in the Add QSO area displayed as UTC as that is a javascript function which gets its time from the browser / users machine which can vary.

You can of course adjust any element of the code its all available.

closed 2015-08-04 by magicbug