magicbug/Cloudlog #171: BUG: DB migrations folder messed up

core-dd, 2015-08-04:
Hej,
I've just stumble over some missing LOTW user and password field in my installation. I have learned about the migration feature to update DB schemata. Unfortunatly, something seems to be broken under 'Cloudlog/application/migrations/'. There exists a doublet of '005_add_dxcc_enddate .php' '005_add_dxcc_enddate.php' and migration 006 seems to be missing. As I'm not experienced with codeigniter may someone more experienced should sort that out? :)

Thanks in advance,
Jens M0INN

magicbug, 2015-08-04:
Hi Jens,

Indeed quite a few bits of been broken over the last year or two.. as things have been tweaked and missed (I'm not installing CL clean that often) 

To fill in what should be in "config"  is a dump below, I really need to spend some time improving the install features, but at the same time also really want to try sorting out a rewrite using Laravel as its become a bit of a nightmare to maintain on this older version of CodeIgniter. 

...
CREATE TABLE IF NOT EXISTS `config` (
  `id` int(9) NOT NULL AUTO_INCREMENT,
  `lotw_download_url` varchar(255) DEFAULT NULL,
  `lotw_upload_url` varchar(255) DEFAULT NULL,
  `lotw_rcvd_mark` varchar(1) DEFAULT NULL,
  `lotw_login_url` varchar(244) DEFAULT NULL,
  `eqsl_download_url` varchar(244) DEFAULT NULL,
  `eqsl_rcvd_mark` varchar(1) DEFAULT NULL,
  PRIMARY KEY (`id`)
) ENGINE=MyISAM  DEFAULT CHARSET=utf8 AUTO_INCREMENT=2 ;
...

core-dd, 2015-08-04:
Hi Peter,
thank you for the short notice. I've deleted '005_add_dxcc_enddate .php' and called index.php/migrate/index to run the migration hook, worked like a charm. I'm not sure whether the migration thing will be called by the installation as I used the supplied *.sql file and edited the corresponding config files manually.

magicbug, 2015-08-05:
no problem, if you fancy pushing them to Git I'd happily merge

magicbug, 2016-01-06:
I've updated .sql files and the one included in the install script.

closed 2016-01-06 by magicbug