magicbug/Cloudlog #174: Error when using same email address on different user accounts

bakoontz, 2015-12-16:
If I'm working with 3 different callsigns, I really don't want to go through the hassle of entering a different email for each.  Probably not a good idea to use the email as a primary key in the table.  Callsign should be sufficient.

magicbug, 2016-01-06:
Email address isn't the primary key of the table? its an indexed field, the primary key is USER_ID.

magicbug, 2016-01-06:
Cloudlog doesn't support multi user logbooks as it states all QSOs are placed in one table and shown on all logins

magicbug, 2016-01-06:
The email was set to unique based on a feature I was going to add but decided not too, you can easily remove the index if that doesn't fit to your use case.

closed 2016-01-06 by magicbug

bakoontz, 2016-01-06:
> Cloudlog doesn't support multi user logbooks as it states all QSOs are placed in one table and shown on all logins 

OK I must have missed that in the docs.  Doesn't do me much good then...