magicbug/Cloudlog #177: PHP 7 compatability

core-dd, 2016-01-11:
Hej hej,
updated to PHP 7 and running into issues now. Start page returns:
'''
A PHP Error was encountered
Severity: Notice
Message: Only variables should be assigned by reference
Filename: core/Controller.php
Line Number: 51
'''
Looks like CodeIgniter is using some depricated statements. Any suggestions besides downgrading PHP?

Cheers,
Jens

magicbug, 2016-01-11:
Current version of codeigniter that cloudlog uses isn't php7 ready, so
sadly the only option is to downgrade for now.

I've not checked how badly the latest codeigniter update breaks the system
yet.

Peter

core-dd, 2016-01-16:
I'm on my way in merging the latest CodeIgniter updates up to 2.2.6. Have to add a database migration file for the minor changes in CIs databases. Migration up to 3.x series is a more advanced endeavour...

magicbug, 2016-01-16:
Yeah I have a sneaky feeling 3x might require some large rewrites in places
but 2.x should in theory merge up ok

core-dd, 2016-02-09:
I've updated my local instance to CI 2.2.6. It's working so far under PHP 5.X without any noticible change. Unfortunatly CI seems to support PHP 7 only in the 3.X branch.

magicbug, 2016-02-09:
Yeah, switching to 3x would require near on rewrite i think, and by then might as well start considering my plans for cloudlog v2 :)

wvolz, 2017-12-01:
I have a working new install of Cloudlog running on CI 3.1.6 and PHP7. I haven't tested the code to upgrade older databases yet, but will try to do that soon. Pull request will come shortly unless you would like me to post a link to my fork.

magicbug, 2017-12-01:
Pull would be fantastic @wvolz sadly development halted my end due to health issues and moving.

wvolz, 2017-12-01:
Pull request #206 created.

magicbug, 2017-12-05:
Thanks @wvolz soon as I get time to test i'll merge

magicbug, 2017-12-13:
Thanks @wvolz Cloudlog is now PHP7 ready.

PR #206 (2017-12-13): Codeignitor 3.1.6 and PHP7 compatibility
This pull request moves the project to Codeignitor 3.1.6 and adds support for PHP7. This should resolve #177 .
closed 2017-12-13 by magicbug