magicbug/Cloudlog #180: null object error when importing direct from eQSL

magicbug, 2016-01-20:
Received via Facebook from someone.

Cloudlog  - Don't know what the ramifications are (I've not looked) but I was getting a null object error when importing direct from eQSL, so I changed the following function...
```
	// Get the last date we received an eQSL
	function eqsl_last_qsl_rcvd_date() {
    	$this->db->select("DATE_FORMAT(COL_EQSL_QSLRDATE,'%Y%m%d') AS COL_EQSL_QSLRDATE", FALSE);
    	$this->db->where('COL_EQSL_QSLRDATE IS NOT NULL');
   		$this->db->order_by("COL_EQSL_QSLRDATE", "desc");
    	$this->db->limit(1);

    	$query = $this->db->get($this->config->item('table_name'));
    	$row = $query->row();

      if(isset($row->COL_EQSL_QSLRDATE))
     		return $row->COL_EQSL_QSLRDATE;
      else
        return null;
  	}
```

magicbug, 2016-01-20:
@skweeker any idea if this will be OK to update, don't personally use eQSL to test it!

teabreakninja, 2016-01-22:
I use eQSL and it's working ok without the extra lines? Was it an "eQSL import" and "Pull eQSL data for me"?
Would I be right in saying $row->COL_EQSL_QSLRDATE would be null if the query returns none, which would happen if all COL_EQSL_QSLRDATE are NULL?

skweeker, 2016-01-22:
Yes @teabreakninja, this would only be an issue if there were only NULL values in that column. I don't think you'd want to return null, though, with out taking a look at the function that calls this one.

Been a while since I've looked at the code (dang kids) but IIRC we're using this function to determine how far back to query eQSL for, when we us the "pull eQSL data for me". You'd want to take a look at that function as well, to see how it'd handle null. If nothing else, have this return some arbitrary date a long, long time ago. I'd say epoch, but there could be records on eQSL older than that :)

Importing an ADIF from eQSL should get this up and running, though, and it should work next time.

magicbug, 2016-01-23:
I'm guessing based on this its a non issue and theres no real need to change anything?

teabreakninja, 2016-01-24:
@magicbug I did have a few Null errors when importing the ADIF file from eQSL so perhaps the process needs a lookover? If it were me, I'd be checking all SQL queries for valid returns, so I'd say your code change above can only help!

If you want, I'll review the whole eQSL process and see if there's anything amiss?

magicbug, 2016-01-24:
@teabreakninja if you wouldn't mind that would be great

magicbug, 2016-01-26:
Merged up #181 Pull as the code looks compatible, thanks @teabreakninja  :+1: for looking at this for me.

closed 2016-02-11 by magicbug