magicbug/Cloudlog #196: ADIF Export Frequency

core-dd, 2016-03-27:
Hej hej,
according to ADIF 3.0.4 (http://www.adif.org/304/ADIF_304.htm) the FREQ field contains the frequency in MHz. Right now export seems to be in Hertz. We might need to change the export for compatibility reasons (e.g. TQSL).

73s de Jens, M0INN

magicbug, 2016-03-31:
Correct Jens the ADIF export needs a tweak to be MHz.

magicbug, 2016-04-02:
Tagged wrong code commit, but now doing hertz / 1000000 to generate correct MHz for ADIF.

magicbug, 2016-04-02:
Closer inspection its for things like 14081011 aka 14.081011 LoTW doesn't like that so throws an error

core-dd, 2016-05-12:
I'll have a look the next time I'm exporting for LOTW. Last time I've manually fixed the exported files via search and replace. ;)

magicbug, 2016-05-14:
It might actually be semi working in the latest codebase, I really need to sit down and start thinking about Version 2 and finding a suitable system to support PHP7 and beyond :)

core-dd, 2016-05-16:
Are you around for the upcoming Camb-Hams trip? We could do a small session on Cloudlog NG ideas. :)

magicbug, 2016-05-16:
I will be 😀

Peter

closed 2016-10-28 by magicbug