magicbug/Cloudlog #199: after installation i got the a few Api problems. the NoApiKey and the MissingKeyMapError

jkoolschijn, 2016-06-26:
Hi, 

after is installed the software on an linux based unit i am getting these two error's

util.js:211 Google Maps API warning: NoApiKeys https://developers.google.com/maps/documentation/javascript/error-messages#no-api-keys
js:32 Google Maps API error: MissingKeyMapError https://developers.google.com/maps/documentation/javascript/error-messages#missing-key-map-error

i can log in and make logs so far so good. but at the dashboard i see the world popping up and then i get these errors ? is there an sollution to this?

magicbug, 2016-06-27:
I'm aware of this Maps API code in the html layouts need updating to reflect change in Google Maps API since it was implemented.

b4sh, 2016-09-01:
This is not a bug. I have gmaps on different domains. On some domains I need to add my API key. I think that soon the key will be required on all domains.

magicbug, 2016-09-02:
Indeed, its not a bug, however code needs updating its just the best way to label it.

lnitram, 2016-10-12:
Found the following places where the key needs to be added:
```
application/views/layout/header.php
application/views/layout/mini_header.php
application/views/view_log/qso.log
```

Would be nice to have a config-entry for this that is used everywhere

PR #216 (2018-11-08): Add configuration parameter for GMaps API key
Hi,

this PR adds a configuration parameter for the GMaps API key, this fixes #199. 

$config['gmaps_api_key'] = "";
can now be set. 

(I'm sorry about the double commits in this PR; I messed up my local branches. This fixes itself if the other PR is merged, otherwise I can rebase this).

So long
Tobias
closed 2018-11-08 by magicbug