magicbug/Cloudlog #201: Clear SAT data on mode and/or band change

m1bxf, 2016-11-07:
Many times I log QSOs made after SAT[ellite] QSO(s) as SAT as the SAT data is not cleared. The SAT data is entered into a TAB not normally visible so it's not till after logging a QSO is it noticed, which then required editing the QSO.

Please could the SAT data be reset (cleared) on a mode or band change? Band might be the better option as some users might change modes (SSB to CW as an example) while working a linear sat.

Raised as a bug due to the undesired behavior.

magicbug, 2018-11-26:
The whole idea is if satellite contact is logged that it will hold the basic details to save time, so this isn't a bug as such, I could set it so that if for sample "SatPC32" isn't the radio selected that the fields are cleared or if the bands changed from 2/70 but really there's no perfect solution

m1bxf, 2018-11-26:
I agree Pete,

I think clearing the sat fields if the band is changed to something not 2m
or 70cm would be the best solution.

If course you could have an option to also clear the sat fields if SatPC32
is not the selected radio.

BR,

Gavin Nesbitt, M1BXF.
http://www.geekshed.co.uk

Sent from my Samsung Note 8, sorry for any typos.