magicbug/Cloudlog #204: Google maps not loading when I open de webpage

datakrom, 2017-07-29:
If I startup the homepage I don't get google maps loaded message received is:
Sorry! Something went wrong.
This page didn't load Google Maps correctly. See the JavaScript console for technical details.
In the java console I get error message 
MissingKeyMapError – An API key is not being used
I have made a new key following this link 
https://churchthemes.com/page-didnt-load-google-maps-correctly/#MissingKeyMapError
The question is how do I get the key loaded into the cloudlog software?
Brgds Rob

R4D4R, 2017-10-23:
I'm having the same issue.  Searching through all of the PHP files, I don't see anywhere to put the API key for Google.

magicbug, 2017-12-05:
Closing due to duplicate issue #199

closed 2017-12-05 by magicbug