magicbug/Cloudlog #205: Callbook issues using Chrome and Vivaldi browsers

ben-kuhn, 2017-09-25:
I am having issues with callbook lookups on the Live QSO page when using Chrome or Vivaldi (chromium-based).  Firefox works fine.

If I clear private data or open in an incognito window, the first 3 or 4 calls look up, but after that, lookups fail.  I am experiencing the same results on several machines (All Ubuntu.)  I realize this may be a bit difficult to troubleshoot, so are there any relevant logs I can gather?

ben-kuhn, 2017-09-29:
Just had the same problem on Firefox.  Clearing cookies resolved it.  What is the correlation to cookies and callbook lookups?

magicbug, 2017-09-29:
Believe theres a session time issue when using the API key provided by hamqth but i've not had time to look into it nor fix it.

ben-kuhn, 2017-09-29:
I'm seeing it with hamqth and QRZ.  Now that I know clearing cookies fixes it it's less critical for me.  Anything I can do to help, logs to gather, etc... that would lift some of the burden let me know.

magicbug, 2017-09-29:
If you see it with QRZ then you could adjust the session/cookie info within https://github.com/magicbug/Cloudlog/blob/master/application/config/config.sample.php

`$config['sess_cookie_name']    = 'ci_cloudlog';
$config['sess_expiration']    = 0;
$config['sess_expire_on_close']  = FALSE;
$config['sess_encrypt_cookie']  = FALSE;
$config['sess_use_database']  = FALSE;
$config['sess_table_name']    = 'ci_sessions';
$config['sess_match_ip']    = FALSE;
$config['sess_match_useragent']  = TRUE;
$config['sess_time_to_update']  = 300;
/*
|--------------------------------------------------------------------------
| Cookie Related Variables
|--------------------------------------------------------------------------
|
| 'cookie_prefix' = Set a prefix if you need to avoid collisions
| 'cookie_domain' = Set to .your-domain.com for site-wide cookies
| 'cookie_path'   =  Typically will be a forward slash
| 'cookie_secure' =  Cookies will only be set if a secure HTTPS connection exists.
|
*/
$config['cookie_prefix']  = "";
$config['cookie_domain']  = "";
$config['cookie_path']    = "/";
$config['cookie_secure']  = FALSE;`

closed 2017-12-05 by magicbug