magicbug/Cloudlog #207: Pull on LoTW not working

mfellman, 2017-12-06:
Not sure if it is config related but moved the site to another server and now the LoTW import (PULL) is not working (If I use the URL it loads the file which looks readable.....):

Severity: Warning

Message: file_get_contents(https://p1k.arrl.org/lotwuser/lotwreport.adi?login=pe1rbr&password=*******&qso_query=1&qso_qsl='yes'&qso_qslsince=2016-10-18 00:00:00&qso_owncall=PE1RBR): failed to open stream: HTTP request failed! HTTP/1.1 400 Bad Request

Filename: controllers/lotw.php

Line Number: 142

Error: Adif_Parser Already Initialized or No in ADIF File

LoTW ADIF Information
A PHP Error was encountered
Severity: Notice

Message: Undefined variable: lotw_table_headers

Filename: lotw/analysis.php

Line Number: 4

A PHP Error was encountered
Severity: Notice

Message: Undefined variable: lotw_table

Filename: lotw/analysis.php

Line Number: 5

magicbug, 2017-12-06:
Just saw this as well looks like the URL to pull ADIF from has changed.

wvolz, 2017-12-13:
The pull request I just submitted should resolve this for new installs. I'm not sure of the best way to resolve this for current installs. You'll need to update your database with the correct URL.

magicbug, 2017-12-13:
Merged the PHP7 and LoTW fixes so if you use the current code should be OK.

Thanks @wvolz for the hard work really appreciate it.

closed 2017-12-13 by magicbug