magicbug/Cloudlog #213: Eqsl errors in download may not be handled correctly

wvolz, 2018-01-05:
I have 2 logbooks @ eqsl and when I login I have to select which one I want. If I try to download content from eqsl into Cloudlog I see a blank screen. When I added debug code it shows the following in the $input string back from CURL:

```
<H3>Error: Multiple users with same Username/Password</H3>
If you change the passwords so they uniquely identify each callsign, or add a QTHNickname parameter, this will not be a problem.
```
Is there a better API to use for downloading eQSL data? It seems like this error should be shown to the user if it occurs.

magicbug, 2018-11-26:
@teabreakninja Any any thoughts? I don't use eQSL to know whether this is still an issue?

teabreakninja, 2018-11-26:
Can't test this as I only have one account, but... "If you change the passwords so they uniquely identify each callsign". Errm What? So, there's the ability to have two accounts, both with the same user/pass combination? and then trying to auth against that throws an error? Seems a bit messed up to me!

magicbug, 2018-11-26:
It looks like its possibly "QTHNickname" allows you to have multiple locations? I can't see anything beyond the following 

> URL to obtain the date/time of last upload of logged-in user (you can specify Username(callsign), Password, and optionally a QTHNickname to find this information without logging in)

Guess we could expose a text box in the user profile area and let users figure out what its meant to mean.

g0wfv, 2018-11-29:
Confirmed!  I have multiples QTHs defined on eQSL and have added code at line 131 in application/controllers/Eqsl.php to temporarily solve the problem (after the eQSL username and password are added to the querystring).

Basically you need to add ...
`&QTHNickname=<Your_QTH_QTHNickname_Here>`

I guess it means a new field in the db table to store the QTHNickname and then add the above to the query string if it exists ....

g0wfv, 2018-11-29:
Add the user_eqsl_qth_nickname column to the users table ....

`alter table users add user_eqsl_qth_nickname varchar(32);`

Git diff patch file (one day I'll fork the repo so I can do PRs!) to add the column to the user editor and use it when downloading the inbox from eQSL.cc ...
 
```
diff --git a/application/controllers/Eqsl.php b/application/controllers/Eqsl.php
index 768e93c..0852b6d 100644
--- a/application/controllers/Eqsl.php
+++ b/application/controllers/Eqsl.php
@@ -106,9 +106,10 @@ class eqsl extends CI_Controller {
 			
 			// Get credentials for eQSL
 			$query = $this->user_model->get_by_id($this->session->userdata('user_id'));
-    		$q = $query->row();
-    		$data['user_eqsl_name'] = $q->user_eqsl_name;
+			$q = $query->row();
+			$data['user_eqsl_name'] = $q->user_eqsl_name;
 			$data['user_eqsl_password'] = $q->user_eqsl_password;
+			$data['user_eqsl_qth_nickname'] = $q->user_eqsl_qth_nickname;
 			
 			// Get URL for downloading the eqsl.cc inbox
 			$query = $query = $this->db->query('SELECT eqsl_download_url FROM config');
@@ -128,6 +129,11 @@ class eqsl extends CI_Controller {
 			$eqsl_url .= "?";
 			$eqsl_url .= "UserName=" . $data['user_eqsl_name'];
 			$eqsl_url .= "&Password=" . $data['user_eqsl_password'];
+
+			if ($data['user_eqsl_qth_nickname'] != '')
+			{
+				$eqsl_url .= "&QTHNickname=" . $data['user_eqsl_qth_nickname'];
+			}
 			
 			$eqsl_url .= "&RcvdSince=" . $eqsl_last_qsl_date;
 			
diff --git a/application/controllers/User.php b/application/controllers/User.php
index fc720cd..44a9f7a 100644
--- a/application/controllers/User.php
+++ b/application/controllers/User.php
@@ -214,6 +214,12 @@ class User extends CI_Controller {
 				$data['user_eqsl_password'] = $q->user_eqsl_password;
 			}
 			
+			if($this->input->post('user_eqsl_qth_nickname')) {
+				$data['user_eqsl_qth_nickname'] = $this->input->post('user_eqsl_qth_nickname');
+			} else {
+				$data['user_eqsl_qth_nickname'] = $q->user_eqsl_qth_nickname;
+			}
+			
 			$this->load->view('user/edit', $data);
 			$this->load->view('layout/footer');
 		}
diff --git a/application/models/User_model.php b/application/models/User_model.php
index a0444ba..ee4fd03 100644
--- a/application/models/User_model.php
+++ b/application/models/User_model.php
@@ -124,7 +124,8 @@ class User_Model extends CI_Model {
 					'user_lastname' => $fields['user_lastname'],
 					'user_timezone' => $fields['user_timezone'],
 					'user_lotw_name' => $fields['user_lotw_name'],
-					'user_eqsl_name' => $fields['user_eqsl_name']
+					'user_eqsl_name' => $fields['user_eqsl_name'],
+					'user_eqsl_qth_nickname' => $fields['user_eqsl_qth_nickname']
 				);
 	
 				// Check to see if the user is allowed to change user levels
@@ -213,14 +214,15 @@ class User_Model extends CI_Model {
 		$u = $this->get_by_id($id);
 
[... 40 lines cut ...]
 		<td><input type="radio" name="lotwimport" id="fetch" value="fetch" /> Pull LoTW data for me</td>
 		<td>
-			<p>Cloudlog will use the LoTW username an password stored in your user profile to download a report from LoTW for you. The report Cloudlog downloads will have all confirmations since your last LoTW confirmation, up until now.</p>
+			<p>Cloudlog will use the LoTW username and password stored in your user profile to download a report from LoTW for you. The report Cloudlog downloads will have all confirmations since your last LoTW confirmation, up until now.</p>
 		</td>
 	</tr>
 </table>
diff --git a/application/views/user/edit.php b/application/views/user/edit.php
index f55a3ef..1e537d2 100644
--- a/application/views/user/edit.php
+++ b/application/views/user/edit.php
@@ -115,6 +115,12 @@ $this->load->helper('form');
 		<?php } ?>
 	</tr>
 	
+	<tr>
+		<td>eQSL.cc QTH Nickname</td>
+		<td><input type="text" name="user_eqsl_qth_nickname" value="<?php if(isset($user_eqsl_qth_nickname)) { echo $user_eqsl_qth_nickname; } ?>" />
+		</td>
+	</tr>
+	
 </table>	
 <input type="hidden" name="id" value="<?php echo $this->uri->segment(3); ?>" />
 <div class="actions"><input class="btn primary" type="submit" value="Update profile" /></div>
```

teabreakninja, 2018-11-30:
Was away with work, I'll get a look at this today...

g0wfv, 2018-11-30:
PR #225 submitted ^^^^

teabreakninja, 2018-11-30:
Yup, looks good. I don't have another account to test the qth_nickname, but the existing code still works!

commit 282a099a (2018-11-30): Merge pull request #225 from g0wfv/develop

Fix #213: Eqsl errors in download may not be handled correctly
commit 9b6faf23 (2018-11-30): Fix #213: Eqsl errors in download may not be handled correctly
closed 2018-11-30 by magicbug