magicbug/Cloudlog #217: [Request for comments] use UTC time for "daily QSO"-counter

Manawyrm, 2018-11-19:
Hi,

I'm in UTC+1 time at the moment, was logging a couple of QSOs at around midnight, and it was showing me "You have 0 QSOs" at 23:01 UTC, 00:01 local time. 
Then (even after completing another QSO) it was of course still showing 0 QSOs.

I have changed the database request in my version of Cloudlog to request the data in UTC from the database as this makes more sense for me. 
Another way to go about this issue would probably be to set PHPs timezone globally to UTC. My server is running on UTC+1 (aka German local time) as well. But I'm not sure if this has other side effects (maybe positive ones?)

Does that behaviour make sense to you? If yes, I'd create a pull request to make this change in the upstream.. :)

vy 73
DO4TMA / Tobias

magicbug, 2018-11-19:
Hi Tobias, yes that makes total sense to me, I run my servers in UK local time which can be UTC or UTC+1

Please create a PR!

Manawyrm, 2018-11-19:
Hi,

a (non-intrusive) commit to fix the display looks like this: 
https://github.com/Manawyrm/Cloudlog/commit/f8932d5b6f1c7e83c6632eb97fd03c1cbd5ca77b
That could be merged in any case, because it just uses the UTC timestamp.

The Logbook model also has other calls to date() without any timestamp reference. 
I'm beginning to think that it might be a good idea to set 
`date_default_timezone_set ( "UTC" );`
in the configuration of the software. 

I think I will test that in my own instance and report back how it works out. 
Do you see any problems with this approach straight-away? 

Thanks
Tobias

magicbug, 2018-11-19:
Thinking about it, within the config.php

```
/*
|--------------------------------------------------------------------------
| Master Time Reference
|--------------------------------------------------------------------------
|
| Options are 'local' or any PHP supported timezone. This preference tells
| the system whether to use your server's local time as the master 'now'
| reference, or convert it to the configured one timezone. See the 'date
| helper' page of the user guide for information regarding date handling.
|
*/
$config['time_reference'] = 'local';
```

Wonder how things behave if you set this to UTC? sorry brains a bit fuzzy today, too many seizures, not enough sleep and a sick pup.

Manawyrm, 2018-11-19:
Good call, I had a look at that too. 
That unfortunately only changes the behaviour of CodeIgnitors own Date commands, not the date()-call for PHP. 

```
$config['time_reference'] = 'UTC';
date_default_timezone_set ( $config['time_reference'] );
```

I did this in my config.php
This seems to work pretty well for me so far. 

No rush, give your dog some treats and sleep well ;)

magicbug, 2018-11-22:
I've set the default timezone to UTC in the configs

magicbug, 2018-11-22:
Do  you think that a suitable thing or further changes needed?

Manawyrm, 2018-11-22:
Looks nice! I've been using this code since Monday and haven't seen any negative effects yet. 
Thanks for patching this :) 

I'll close this issue :)

closed 2018-11-22 by Manawyrm