magicbug/Cloudlog #222: FEATURE: Remove Tools from Menu replace with just Awards

magicbug, 2018-11-26:
I'm proposing that we remove Tools from the top menu, I don't think theres much need for having DX Cluster so that code can be removed and just replace it with Awards?

teabreakninja, 2018-11-27:
Never used the DXCluster this way myself. Maybe other people use it though?

magicbug, 2018-11-27:
I wouldn't have thought so, theres far better cluster websites even mine at dxcluster.co.uk or dxheat, what I might do is remove it, see if anyone complains then later on scrap the dx cluster code

magicbug, 2018-11-27:
[https://github.com/magicbug/Cloudlog/commit/e9893d8907ae23420818fb9ea6bf3ee74982cc1f](https://github.com/magicbug/Cloudlog/commit/e9893d8907ae23420818fb9ea6bf3ee74982cc1f) I've now commited this will leave the issue open for a while in case anyone really objects

closed 2018-11-27 by magicbug