magicbug/Cloudlog #223: Map issue on /application/views/view_log/qso.php

magicbug, 2018-11-27:
@teabreakninja there appears to be an issue with the map or perhaps a caching issue, normally when clicking on a callsign it would show the map centred on the user with a pin.

This doesn't seem to happen any more just zooms in but no pin, however if i zoom out I can see all the QSOs which it shouldn't.

teabreakninja, 2018-11-27:
[image]
Is this what you mean? This works for me.

magicbug, 2018-11-27:
Yes, should only be the one pin for the contact in question

I either get two pins guess its getting it from the maps list the front page uses
[image]

Or if its beyond the time cut off for the front page, it zooms into the area but no pin
[image]

This is what it used to do on Google Maps

[image]

I tend to use these screens for grabbing screenshots for tweets.

magicbug, 2018-11-27:
Theres actually an issue on /index.php/logbook/ where the map should only show the QSOs for whats displayed on the screen where as again its showing the information from the home screen map

magicbug, 2018-11-27:
OK I've updated the map code on the QSO popup can be seen here https://github.com/magicbug/Cloudlog/commit/906a4ce249ba60132fc900e34fa804b37b3cc551

What I haven't done yet is the logbook one

magicbug, 2018-11-27:
Fixed logbook aswell https://github.com/magicbug/Cloudlog/commit/1690bd1bb8e19296ce14ddad3fd3903138abf573

closed 2018-11-27 by magicbug