magicbug/Cloudlog #229: Add QSO: QSO details remain after deletion of callsign field.

g0wfv, 2018-12-07:
**Describe the bug**
If you start to enter a QSO by typing in the callsign, certain field will be populated by lookup and you can also add other fields manually.

If you then decide not to log the QSO (you didn;t make the contact perhaps?) and delete the callsign, the previously entered callsign's details (Name and Locator eg.) are still populated and will not update on entry of the next callsign. 

**To Reproduce**
Steps to reproduce the behavior:
1. Log a QSO with the name field populated.
2. Add a second QSO with the name field populated.
3. Type the first callsign into the callsign field (name field - and others) populated by lookup_.
4. Remove the callsign from the callsign field.
5. Type the second callsign into the callsign field (previously lookup data persists).

**Expected behavior**
The QSO data fields (except static data not pertaining to an individual station) should be removed on deletion of the callsign field as though you were adding a "fresh" QSO.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Linux
 - Browser Chrome
 - Version 70.0.3538.77 (Official Build) (64-bit)

**Additional context**
I'm looking into it ... (bare with me might not be within my skills to sort out the logic in PHP!)

magicbug, 2018-12-07:
Indeed, although pressing reset also clears all fields, could use the JavaScript keyup and watch it the input does go fully blank

g0wfv, 2018-12-08:
Javascript, PHP, it's all "new" to me ..... I am learning though :-)

g0wfv, 2018-12-08:
... but not quick enough!

Spent the last few hours scratching my head over this one (on and off!)

keyup does indeed seem to be the way to go (after introducing a delay to wait for typing to stop else you get way too many db lookups when they're not required!)

However, I just can't get the QSO list to update back to the full list when the callsign input goes empty (ie with no filter) despite trying various ways to pass the '%' catch all through to Logbook.php or other workaround hacks.

magicbug, 2018-12-09:
If I remember right, the last 16 qsos was hard coded and not loaded by JS, so it would need a similar function to the search to make it return.

g0wfv, 2018-12-09:
I'm making some progress now.

Trying to pass a % back through was a red herring (and possibly being stopped by the webserver for security reasons anyway?)

I am wondering why the auto and post QSO adding uses different files?  I've temporarily copied index.php (with my changes) to manual.php and added more changes to that so it works with the extra URL path additions, and all seems to be fine; I've not noticed anything vastly untoward.  It seems to me the two files should be the same with the exception of updating the time and if connected via CAT, the frequency/band/mode automatically?  Currently they aren't too dissimilar, but do accomplish things in slightly differing ways (I'm wondering if the two files have diverged over time?)

Perhaps a more efficient way would to have a checkbox that enables and disables the automatic features on a single form thus decreasing the need to maintain two similar (but not the same!) files.

Incidentally, the 16 last QSOs is hard coded (good memory) but only for the initial view; for some reason it changes to 10 once you've added a QSO (the partial search remains at 16 max).

I'm slowly working my way through making changes as I find things - it's good for learning!

magicbug, 2018-12-09:
Live & Post have different interfaces because of the logic is slightly different, in Post your back dating the qso most likely, whereas live its "now" I didn't want buttons to like HRD worked its ugly having to press start and finish on a time. Yes Radio functions are also not provided live in the Post (Manual) entry area.

magicbug, 2018-12-12:
Did you get any further with this? Or should I take a look

g0wfv, 2018-12-12:
Sorry, yes, I did - I even folded it all into one php file and used a query string to alter the behaviour dependent on live or post logging.  But I need to check the logic differences (CAT etc) between the two files and amend as appropriate whilst testing - I've not yest tried the CAT functions.

Suffice it to say it's working quite nicely (although there's probably room for improvement once someone eho really knows what they're doing gets hold of the code!)  I shall close it now, as I will submit this upstream when I've bedded it in some more ....

closed 2018-12-12 by g0wfv