magicbug/Cloudlog #238: PHP socket call to rigctld

magicbug, 2018-12-17:
As most are aware CloudlogCAT is now working for Windows and its Omnirig based, however of course we can't forget linux etc.

Therefor rigctld is purposed, haven't got a demo yet, but will use this for discussion.

g0wfv, 2018-12-17:
Just thinking aloud - you could keep it simple for Cloudlog and use the PHP call to rigctld as I mentioned; the only problem would be (as mentioned by @Manawyrm) the hosting of Cloudlog outside of the LAN housing the computer hosting rigcltd.  This could be overcome with the use of a VPN rather than trying to get Cloudlog to cover all bases? or an SSH tunnel?

magicbug, 2018-12-17:
Yeah, I host mine outside the LAN too in a commercial data centre, which makes me think it would be better to keep the two separate. I also supply around 25 hosted instances of cloudlog too at the moment for people doing similar and some use Linux hence why I thought keeping it external would make sense

magicbug, 2018-12-17:
The example of CloudlogCAT, it runs on the desktop and just does a json post when rig changes a value, this is similar for SatPC32.

g0wfv, 2018-12-17:
Sorry, I didn't understand the mechanics of this vefore you said that; so
Cloudlog is already capable of accepting a JSON post to alert it to the
frequency and mode?  All you need in that case, is a script to do so that
interfaces with rigctld?  I already have just the thing then ..... well
close enough; I wrote a shell script that runs as a daemon process and
syncs my rigctld controlled radio with my RTL-SDR (acting as a panadapter)
via gqrx's remote control - just replace the gqrx remote call with a json
post from cURL and we''re good to go ..... unfortunately it's on my laptop
..... which is at home (I'm currently working away on another laptop until
tomorrow evening!)  Give me a clue how to work the JSON post, and I can do
the modifications and test it in my time off from Wednesday until the
weekend ......

> --

A J Corbett
G0WFV

Manawyrm, 2018-12-17:
Hi, 

yes, that's exactly it.
A quick curl call in your shell script should do the trick.

I'm going to upload my PHP implementation in just a minute, will want to clean it up just a tiny bit. 

I was a bit surprised that (as far as I can see) nobody has written a PHP rigctld socket implementation yet. Everybody seems to be calling the rigctl binary directly, which is a bit dirty IMHO. 
I've implemented the network protocol for rigctld and it's using curl for the POST, so it should even work on Windows. (if you're willing to use a PHP shell on windows). 

Will report back with my code, then you can copy the JSON call.

Manawyrm, 2018-12-17:
Here's my implementation: 
https://github.com/Manawyrm/cloudlog-rigctl-interface/tree/master

It's a bit hacky, but at least it's not using shell_exec() but the TCP socket instead.
@g0wfv Take a look at rigctlCloudlogInterface.php, there's the array that needs to be sent to the API endpoint. 

@magicbug I'm a bit worried about the "open" API endpoint for the radio API. I'm thinking about adding an optional API key. Just hardcoded in the config. Would a PR for this be accepted? 
Could be backwards compatible with the current scripts if no key is defined.

magicbug, 2018-12-17:
Key system is there in api section, just need to add check it's valid I'm adding that in the coming days.

magicbug, 2018-12-17:
That said @Manawyrm if you have time to do it feel free.

My thinking is used created API key via the current system and its just included in JSON post

Manawyrm, 2018-12-17:
OK, I'll implement that using the current system and send a PR :)

g0wfv, 2018-12-17:
How do you run that PHP without webserver? (I'm a complete noob to PHP BTW!)

magicbug, 2018-12-17:
PHP runs on the command line :) if you have the latest it's also got a built in server.

Manawyrm, 2018-12-17:
> How do you run that PHP without webserver? (I'm a complete noob to PHP BTW!)

Thanks for asking, good point. 
I have added some more documentation to the readme file: 
https://github.com/Manawyrm/cloudlog-rigctl-interface/blob/master/README.md
(and it now also supports API with key authentication, as mentioned in the other pull request)

Is this clearer now? Any recommendations for additional documentation? 

I'm a bit biased there, as a PHP coder by trade I'm probably missing some details for beginners. Sorry about that :-)

g0wfv, 2018-12-17:
Screen or tmux (if like me you're odd!)

Also a good old shebang will also make it run from the command line too (apparently!)

``#! /usr/bin/php`` or similar then just ``chmod +x`` the file and Bob's your uncle ...

Also rigctld can accept more than one command at a time, so ``fm`` (if I recall correctly) will return freq, mode and bandwidth in one query/response instead of 2 (if you were in to optimising for speed!)

Manawyrm, 2018-12-17:
Very nice info, thanks!
I've added the shebang and changed the request command to fm. This might be helpful, especially with older radios with 4800baud CAT control. Not sure how efficiently rigctl handles this internally (my radio luckily does 115200 baud)...

g0wfv, 2018-12-17:
I could, of course, be talking complete crap as I haven't got my script to hand to confirm anything until I get home tomorrow evening - I do recall that one of either rigctld or gqrx accepted multiple commands at once .... I'm 95% certain it was rigctld but what I'm not sure about was whether the command could be sent as ``fm`` or if it had to have a space - ``f m`` - perhaps you already tested it anyways?

Manawyrm, 2018-12-17:
No, you're right. It works fine with fm ;) 
I've already tested and implemented it into my code... Works!

g0wfv, 2018-12-17:
Oh that's excellent!  Another thing to test is if it will run in the background without screen or tmux by running it ....

``./rigctlCloudlogInterface.php & >/dev/null 2>&1`` or perhaps``nohup ./rigctlCloudlogInterface.php & >/dev/null 2>&1``

magicbug, 2018-12-17:
Merged into the main branch, will update my two applications tomorrow to include the API bits :)

Manawyrm, 2018-12-17:
I think this answer of yours was meant for the other thread... The 2 PRs and 1 issue is a bit chaotic... :smile: 

But this issue can now probably be closed? 

btw: Thanks for the shout-out in the readme :+1:

magicbug, 2018-12-17:
Most welcome, tbh soo many emails popping up I always get confused lol

right onto the next thing that pops up 👍

closed 2018-12-17 by magicbug

g0wfv, 2018-12-20:
Just for completeness, a (very simplistic) bash script to synch Cloudlog with rigctld ....

[https://github.com/g0wfv/CloudlogBashCat](https://github.com/g0wfv/CloudlogBashCat)