magicbug/Cloudlog #246: 8 or 10 character grid squares are not handled properly

ZigmundRat, 2019-01-10:
If 8 or 10 character grid squares locations are entered, the calculated distance and location information is entirely incorrect. For example FN22EP vs FN22EP10BG - FN22EP10BG is placed 4768.5 miles away.

magicbug, 2019-01-11:
Yes suspect there's a good chance it doesn't, as 6 tends to be the longest given on air or used in the hobby I've only development with that in mind.

However I'll see if it's a quick fix

ZigmundRat, 2019-01-11:
Sounds good, thx. You could  try to find the actual lat/lon of the 8/10 character location or just truncate to 6 characters - either is OK with me. 

I also experienced a variety of other issues during the installation and initial configuration. Would you like me to open issues for those as well?

magicbug, 2019-01-14:
I'll code in a trim soon for gridsquares, been working on upgrading the interface to Bootstrap 4

ZigmundRat, 2019-01-18:
OK. I'll wait for the update and do a new install, documenting issues along the way. Thanks!

closed 2019-05-04 by magicbug