magicbug/Cloudlog #249: Discussion: satellite/relay frequency offsets

Manawyrm, 2019-02-24:
While talking to some other OMs on the new satellite, my online logbook came into discussion. 
The frequency input I had done was talked about and someone brought up that it was custom to log the uplink/TX frequency in the logbook. 
This seems to be reinforced by the existence of the COL_FREQ_RX column in the database, which is currently unused. 

With repeaters you often have the -7.6MHz offset, in case of the es'hail it's a +8089,5MHz offset.. 
Any ideas on how/if this could be integrated into the UI? 
Either seperate frequency input fields (those could be filled by the API/CAT integration), or (crude!) maybe just a list of birds, if "sat_name" == es'hail-2 -> offset +8089,5M. 

Is that a useful feature in general? How do you log sat/repeater contacts?

magicbug, 2019-02-24:
I actually send it on my app from SatPC32 it's stored in the database just need to add it to the frontend

magicbug, 2019-02-25:
In the QSO area under "Station" there's the frequency input box, I'm proposing we add Frequency RX below to store the RX this can then be populated manually or by using the API cat system.

[image]

Manawyrm, 2019-02-25:
Good idea. I think I'll have a go at that this evening.

magicbug, 2019-02-25:
Think in the database for CAT/Radios the downlink is stored in a field called "downlink" or something close to it! Just need a little extra js to add it.

magicbug, 2019-02-25:
While thinking about it, I can't remember if there is also fields for band and mode for RX however perhaps should populate those

Manawyrm, 2019-02-25:
Ok, so there's a field for RX band, not for RX mode. That probably makes sense :) 
I'll implemented that later.

Another thing I'd like to add is a better frequency parser for the input field.
"3700" -> 3.700 MHz 
"2350210000" -> 2.35021 GHz
"2.350e6" -> 2.35 GHz
"3.7M" -> 3.700 MHz
What do you think about that? Should be easy to do and not break any existing mechanisms...

magicbug, 2019-02-25:
Yes would like to show the frequencies better a long number is a bit overwhelming to most, it would also look nicer on the display.

magicbug, 2019-02-25:
Merged the changes they look great, also note the change to big int, to handle the larger numbers.

Trying to think if satellite "mode" is handled for QO-100, for example, 2m up and 70cms down is "V/U" not sure if "S/X" is being populated if you use SatPC32 or similar.

magicbug, 2019-02-25:
In fact, just had a look [https://github.com/magicbug/Cloudlog/blob/master/application/controllers/Radio.php](https://github.com/magicbug/Cloudlog/blob/master/application/controllers/Radio.php) go to `satmode()` need to improve that to support all as its missing most the designators can be found at [http://www.pe0sat.vgnet.nl/satellite/sat-information/modes/](http://www.pe0sat.vgnet.nl/satellite/sat-information/modes/)

Manawyrm, 2019-02-25:
> codes duplicated for up and down

100% agree. Especially if you look at my other new commit (with the json), the seperate requests are not useful at all. 
I'll clean that up and add all the modes/frequency ranges later on.

Manawyrm, 2019-02-25:
With #263, this should all work nicely :)

closed 2019-02-25 by Manawyrm