magicbug/Cloudlog #250: https://github.com/magicbug/Cloudlog/blob/master/sql/tables/station_profile.sql

Manawyrm, 2019-02-24:
Hi,

the station_profile table is missing a primary key and an auto-increment flag. 
https://github.com/magicbug/Cloudlog/blob/master/sql/tables/station_profile.sql

`ALTER TABLE `station_profile`
MODIFY COLUMN `station_id`  int(11) UNSIGNED NOT NULL AUTO_INCREMENT FIRST ,
ADD PRIMARY KEY (`station_id`);`

I'd recommend adding this to the file...

So long,
Tobias

magicbug, 2019-02-24:
Whoops not sure why that was missing

Manawyrm, 2019-02-25:
Thought so. Should I open a PR? 
There's another issue in the new BS4 parts, in the API-keys the Link is incorrect.

Can I send a PR for that, too, or are you currently working on these files?

magicbug, 2019-02-25:
No feel free to play with those bits not touching API area at the moment

closed 2019-02-25 by magicbug