magicbug/Cloudlog #259: Fetch callsign info as JSON

Manawyrm, 2019-02-25:
as mentioned in #258, it'd be a good idea to fetch the info for a callsign as a single JSON request, like it's now done for the CAT data.

magicbug, 2019-02-25:
Not sure if its possible to improve the speeds in other ways, another thing I'd love is check when a gridsquare is "new" for a satellite contact and change the box colour to Red (worked) or Green for (new) but that's more nice to haves :)

Manawyrm, 2019-02-26:
With the new changes, displaying the grid-worked status is no problem.
I'm not much of a sattelite OP, how much do the grids have to match? 
is 4-letter grid (JO41) or 6-letter (JO41wx) matching required for a worked display? 

Any color suggestions? I'm a bit colourblind when it comes to design things ;)

magicbug, 2019-02-26:
4 letter grid JO41 is plenty 6 I tend to just use for showing on the map ;-)

"New" #4BB543
"Worked" #B33A3A

magicbug, 2019-02-27:
Nice thats useful, from the code its checking based on all the log, anyway we can narrow it down for HF/2/6/SATS so when making a json call we just include band and perhaps "radio" status?

Manawyrm, 2019-02-27:
Sure, we can include that in the lookup. 
I'm not 100% sure how you would want the feature to work yet. 
 -> Callsign-lookup with Band set to an HF band searches only HF bands, 
 -> Callsign-lookup with Band set to VHF/UHF band searches only VHF/UHF bands
 -> Callsign-lookup with Propagation-mode set to Sat searches only prop.mode==sat QSOs

Is that about right? Maybe we'd need to split this lookup into a second request then. When entering a QSO manually, not via the CAT, the band/prop-mode isn't known yet...

magicbug, 2019-02-27:
Interestingly when Sat mode/name is populated I don't populate the dropdown i just do it in the backend but yes that would make the most sense!

Yes that would be perfect, I guess theres also a case of if you enter the gridsquare after the name etc is populated or change it, it should double check?

magicbug, 2019-06-13:
I made some adjustments as this seemed to use the Clublog API for lookup rather than local database, also now returns lat/long from gridsquare if its available.

magicbug, 2019-08-16:
This now works roughly how i'd expect

When entering a callsign it checks 

- If sat_name is filled in returns based on SAT qsos
- else it returns based on band/mode selection

PR #264 (2019-02-26): Fetch all callsign information in a single JSON request
This fixes #259 and speeds the mechanism up by a lot. 
Now the only delay is the 500ms keyboard delay.

[image]

Name, QTH/QRA, Bearing and past QSOs get passed as a single JSON array: 
[image]

I also cleaned the code up and removed a lot of duplicate code, it doesn't do 4 requests to HamQTH/qrz every time now, but just one.
closed 2019-08-16 by magicbug