magicbug/Cloudlog #260: Thoughts: Add QSO Area

magicbug, 2019-02-25:
Before I start making any switch of the QSO inputting areas to Bootstrap4, I think its probably worth having a think about the area in general.

- Is the title "Add QSO" at the top needed/redundant seems a bit pointless?
- Should the locator box be made smaller to allow information about whether its new/worked to be shown beside it, along with some logic so it would only show based on the style of contact like "HF" or "Sats" would not pool from the same set of contacts?

I'm not really looking at allowing custom layouts for this page, think if people want that I'll leave it up to them to completely rewrite the template for this section, but open to feedback on how the current layout can be and should be improved.

Manawyrm, 2019-02-26:
I only have one small change proposal: I'd swap the tab pages, Station to be first, Home after that.
I never set propagation mode or IOTA, but always input a frequency for the QSOs.

magicbug, 2019-02-26:
This is actually a good point, I only use Propagation mode for MS and Sat QSOs!

Which then options question does home/station tabs perhaps need to be renamed.

magicbug, 2019-03-06:
Another comment was to expose a sota field similar to iota which I have no objection to, it would be nice to have these as auto complete at some stage

magicbug, 2019-05-22:
Those who use the current branch will see the update to the QSO page.

Manawyrm, 2019-05-22:
[image]

Looks great, much more usable on a phone now. :+1:

magicbug, 2019-05-31:
Thanks, certainly feels better to use, in its current state what do people think meds to be changed or added? I'd prefer past QSOs to be a data table with is which I'm going to look into.

I've had a request to show a map for the current QSO, which could be useful for VHF and up.

magicbug, 2019-06-13:
Tim G4VXE asked whether a map could be added to the add qso page, which of course it can, the end result is this. it will either calculate lat/long from typed Gridsquare, from a callbook or if nothing else the DXCC lat/long 

[image]

magicbug, 2019-06-13:
There is also now a SOTA Reference box under "general" this is stored in the SOTA field as per ADIF specs and if you go to the Awards page contacts are shown for record keeping

closed 2019-08-16 by magicbug