magicbug/Cloudlog #261: Thoughts: Dashboard Improvements/Changes

magicbug, 2019-02-25:
With the impending change to the dashboard/home screen, does anyone feel any changes need to be made, more information displayed or even less?

I'm considering hiding QSL Card information to none logged in users on this area, I do wonder whether to add a box to show radios connected via the CAT system?

magicbug, 2019-02-25:
This could be the time to look at custom pins around issue #202 regarding different coloured pins for different bands, satellites, modes, I think this could get tricky though with such a broad selection so might just need to limit to it to bands?

g7kse, 2019-03-06:
A couple of things for me:
1. I'm not sure how to handle the SOTA part but I could appreciate a way of visualising that information. Summits activated, points etc. I don't think I've labelled / tagged mine correctly as they don't appear in the current awards tab
2. Integrate DX cluster or RBN perhaps. 

TBH I quite like it how it is and as long as it stays as clean as it is I'm a fan

magicbug, 2019-03-06:
SOTA needs improvement, it has fields in the Adif spec & the table now just need to make it work in the frontend similar to iota, I might make it so it has to be turned on or off in the QSO window

DX cluster / rbn - for the dashboard do you mean show who's spotted you?

g7kse, 2019-03-07:
Now there's a thought for RBN. That would be really useful. For the cluster I just thought a menu button that would just tie the two apps together. There are probably a ton of things that could actually be added but whether or not they improve cloudlog could be debated e.g qrz.com lookup, /p companion app for off line logging (sync to app)

Probably the biggest turn off for me at the start was the installation, it does take a bit of thinking about and is not too easy for us mortals. Performance on the Rpi zero W is goos and RPI 3B is excellent. Some kind of automated installation would be awesome (probaby a nightmare to do).

I'm full of 'good' ideas ut probably no help in actually being able to achieve anything ;-)

magicbug, 2019-03-07:
Installs improving, least with the current process you can login without having to change configs! but its one process I've not really mastered.

Yes there's always potential to add more, but I take the stance is less is more, with making as much accessible to third party stuff

magicbug, 2019-03-07:
RBN/Spotted on dashboard could be do able, I don't collect RBN not sure if theres an API somewhere for it, should think theres a huge amount of data flowing compared to the usual cluster stuff for dxcluster.co.uk so not sure I want to deal with that lol

g7kse, 2019-03-07:
I agree with the less is more thing. It is really good as it is. Everything else could end making it just like every other 'bloat' log

closed 2019-05-14 by magicbug