magicbug/Cloudlog #267: DB Migration v1.1 to v1.7 missing some fields

core-dd, 2019-02-27:
I've finally managed to reactivate my old install (v1.1 or something) and went through the DB migration process. It seems that at least the `COL_VUCC_GRIDS` field was lost in the migration process.
The following captures my current state of the `TABLE_HRD_CONTACTS_V01`:
```
MariaDB [cloudlog]> show columns from TABLE_HRD_CONTACTS_V01;
+----------------------+--------------+------+-----+---------+----------------+
| Field                | Type         | Null | Key | Default | Extra          |
+----------------------+--------------+------+-----+---------+----------------+
| COL_PRIMARY_KEY      | int(11)      | NO   | PRI | NULL    | auto_increment |
| COL_ADDRESS          | varchar(255) | YES  |     | NULL    |                |
| COL_AGE              | int(11)      | YES  |     | NULL    |                |
| COL_A_INDEX          | double       | YES  |     | NULL    |                |
| COL_ANT_AZ           | double       | YES  |     | NULL    |                |
| COL_ANT_EL           | double       | YES  |     | NULL    |                |
| COL_ANT_PATH         | varchar(2)   | YES  |     | NULL    |                |
| COL_ARRL_SECT        | varchar(10)  | YES  |     | NULL    |                |
| COL_BAND             | varchar(10)  | YES  | MUL | NULL    |                |
| COL_BAND_RX          | varchar(10)  | YES  |     | NULL    |                |
| COL_BIOGRAPHY        | longtext     | YES  |     | NULL    |                |
| COL_CALL             | varchar(32)  | YES  | MUL | NULL    |                |
| COL_CHECK            | varchar(8)   | YES  |     | NULL    |                |
| COL_CLASS            | varchar(8)   | YES  |     | NULL    |                |
| COL_CNTY             | varchar(32)  | YES  |     | NULL    |                |
| COL_COMMENT          | longtext     | YES  |     | NULL    |                |
| COL_CONT             | varchar(6)   | YES  | MUL | NULL    |                |
| COL_CONTACTED_OP     | varchar(32)  | YES  |     | NULL    |                |
| COL_CONTEST_ID       | varchar(32)  | YES  |     | NULL    |                |
| COL_COUNTRY          | varchar(64)  | YES  |     | NULL    |                |
| COL_CQZ              | int(11)      | YES  |     | NULL    |                |
| COL_DISTANCE         | double       | YES  |     | NULL    |                |
| COL_DXCC             | varchar(6)   | YES  | MUL | NULL    |                |
| COL_EMAIL            | varchar(32)  | YES  |     | NULL    |                |
| COL_EQ_CALL          | varchar(32)  | YES  |     | NULL    |                |
| COL_EQSL_QSLRDATE    | datetime     | YES  |     | NULL    |                |
| COL_EQSL_QSLSDATE    | datetime     | YES  |     | NULL    |                |
| COL_EQSL_QSL_RCVD    | varchar(2)   | YES  |     | NULL    |                |
| COL_EQSL_QSL_SENT    | varchar(2)   | YES  |     | NULL    |                |
| COL_EQSL_STATUS      | varchar(255) | YES  |     | NULL    |                |
| COL_FORCE_INIT       | int(11)      | YES  |     | NULL    |                |
| COL_FREQ             | bigint(13)   | YES  |     | NULL    |                |
| COL_FREQ_RX          | bigint(13)   | YES  |     | NULL    |                |
| COL_GRIDSQUARE       | varchar(12)  | YES  |     | NULL    |                |
| COL_HEADING          | double       | YES  |     | NULL    |                |
| COL_IOTA             | varchar(10)  | YES  | MUL | NULL    |                |
| COL_ITUZ             | int(11)      | YES  |     | NULL    |                |
| COL_K_INDEX          | double       | YES  |     | NULL    |                |
| COL_LAT              | double       | YES  |     | NULL    |                |
| COL_LON              | double       | YES  |     | NULL    |                |
| COL_LOTW_QSLRDATE    | datetime     | YES  |     | NULL    |                |
| COL_LOTW_QSLSDATE    | datetime     | YES  |     | NULL    |                |
| COL_LOTW_QSL_RCVD    | varchar(2)   | YES  |     | NULL    |                |
| COL_LOTW_QSL_SENT    | varchar(2)   | YES  |     | NULL    |                |
| COL_LOTW_STATUS      | varchar(255) | YES  |     | NULL    |                |
| COL_MAX_BURSTS       | int(11)      | YES  |     | NULL    |                |
| COL_MODE             | varchar(10)  | YES  | MUL | NULL    |                |
| COL_MS_SHOWER        | varchar(32)  | YES  |     | NULL    |                |
| COL_MY_CITY          | varchar(32)  | YES  |     | NULL    |                |
| COL_MY_CNTY          | varchar(32)  | YES  |     | NULL    |                |
| COL_MY_COUNTRY       | varchar(64)  | YES  |     | NULL    |                |
| COL_MY_CQ_ZONE       | int(11)      | YES  |     | NULL    |                |
| COL_MY_GRIDSQUARE    | varchar(12)  | YES  |     | NULL    |                |
| COL_MY_IOTA          | varchar(10)  | YES  |     | NULL    |                |
| COL_MY_ITU_ZONE      | int(11)      | YES  |     | NULL    |                |
| COL_MY_LAT           | double       | YES  |     | NULL    |                |
| COL_MY_LON           | double       | YES  |     | NULL    |                |
| COL_MY_NAME          | varchar(64)  | YES  |     | NULL    |                |
| COL_MY_POSTAL_CODE   | varchar(24)  | YES  |     | NULL    |                |
| COL_MY_RIG           | varchar(255) | YES  |     | NULL    |                |
| COL_MY_SIG           | varchar(32)  | YES  |     | NULL    |                |
| COL_MY_SIG_INFO      | varchar(64)  | YES  |     | NULL    |                |
| COL_MY_STATE         | varchar(32)  | YES  |     | NULL    |                |
[... 35 lines cut ...]
| COL_STX              | int(11)      | YES  |     | NULL    |                |
| COL_STX_STRING       | varchar(32)  | YES  |     | NULL    |                |
| COL_SWL              | int(11)      | YES  |     | NULL    |                |
| COL_TEN_TEN          | int(11)      | YES  |     | NULL    |                |
| COL_TIME_OFF         | datetime     | YES  |     | NULL    |                |
| COL_TIME_ON          | datetime     | YES  | MUL | NULL    |                |
| COL_TX_PWR           | double       | YES  |     | NULL    |                |
| COL_WEB              | varchar(128) | YES  |     | NULL    |                |
| COL_USER_DEFINED_0   | varchar(64)  | YES  |     | NULL    |                |
| COL_USER_DEFINED_1   | varchar(64)  | YES  |     | NULL    |                |
| COL_USER_DEFINED_2   | varchar(64)  | YES  |     | NULL    |                |
| COL_USER_DEFINED_3   | varchar(64)  | YES  |     | NULL    |                |
| COL_USER_DEFINED_4   | varchar(64)  | YES  |     | NULL    |                |
| COL_USER_DEFINED_5   | varchar(64)  | YES  |     | NULL    |                |
| COL_USER_DEFINED_6   | varchar(64)  | YES  |     | NULL    |                |
| COL_USER_DEFINED_7   | varchar(64)  | YES  |     | NULL    |                |
| COL_USER_DEFINED_8   | varchar(64)  | YES  |     | NULL    |                |
| COL_USER_DEFINED_9   | varchar(64)  | YES  |     | NULL    |                |
| COL_CREDIT_GRANTED   | varchar(64)  | YES  |     | NULL    |                |
| COL_CREDIT_SUBMITTED | varchar(64)  | YES  |     | NULL    |                |
+----------------------+--------------+------+-----+---------+----------------+
118 rows in set (0.00 sec)
```
Hope this helps :)
Thanks for the recent work on Cloudlog!

magicbug, 2019-02-28:
Ah yes it's in the install but not the migration, I added quite a few fields to the log table

magicbug, 2019-02-28:
@core-dd Jens, good to see you back, been enjoying reading the la1ark tweets, you are correct about missing rows for the log table which I added things like VUCC, can you believe they weren't there from the beginning?! I'm ashamed :p

@Manawyrm would you like me to create a migration for these?

```SQL
ALTER TABLE `TABLE_HRD_CONTACTS_V01`  ADD `COL_ADDRESS_INTL` VARCHAR(255) NULL DEFAULT NULL  AFTER `COL_CREDIT_SUBMITTED`,  ADD `COL_AWARD_GRANTED` VARCHAR(255) NULL DEFAULT NULL  AFTER `COL_ADDRESS_INTL`,  ADD `COL_AWARD_SUMMITED` VARCHAR(255) NULL DEFAULT NULL  AFTER `COL_AWARD_GRANTED`,  ADD `COL_CLUBLOG_QSO_UPLOAD_DATE` DATETIME NULL DEFAULT NULL  AFTER `COL_AWARD_SUMMITED`,  ADD `COL_CLUBLOG_QSO_UPLOAD_STATUS` VARCHAR(20) NULL DEFAULT NULL  AFTER `COL_CLUBLOG_QSO_UPLOAD_DATE`,  ADD `COL_COMMENT_INTL` VARCHAR(255) NULL DEFAULT NULL  AFTER `COL_CLUBLOG_QSO_UPLOAD_STATUS`,  ADD `COL_COUNTRY_INTL` VARCHAR(255) NULL DEFAULT NULL  AFTER `COL_COMMENT_INTL`,  ADD `COL_SILENT_KEY` VARCHAR(2) NULL DEFAULT NULL  AFTER `COL_COUNTRY_INTL`,  ADD `COL_SKCC` VARCHAR(255) NULL DEFAULT NULL  AFTER `COL_SILENT_KEY`,  ADD `COL_DARC_DOK` VARCHAR(10) NULL DEFAULT NULL  AFTER `COL_SKCC`,  ADD `COL_FISTS` INT(10) NULL DEFAULT NULL  AFTER `COL_DARC_DOK`,  ADD `COL_FISTS_CC` INT(10) NULL DEFAULT NULL  AFTER `COL_FISTS`,  ADD `COL_HRDLOG_QSO_UPLOAD_DATE` DATETIME NULL DEFAULT NULL  AFTER `COL_FISTS_CC`,  ADD `COL_HRDLOG_QSO_UPLOAD_STATUS` VARCHAR(10) NULL DEFAULT NULL  AFTER `COL_HRDLOG_QSO_UPLOAD_DATE`,  ADD `COL_MY_ANTENNA` VARCHAR(255) NULL DEFAULT NULL  AFTER `COL_HRDLOG_QSO_UPLOAD_STATUS`,  ADD `COL_MY_ANTENNA_INTL` VARCHAR(255) NULL DEFAULT NULL  AFTER `COL_MY_ANTENNA`,  ADD `COL_MY_CITY_INTL` VARCHAR(255) NULL DEFAULT NULL  AFTER `COL_MY_ANTENNA_INTL`,  ADD `COL_MY_COUNTRY_INTL` INT(6) NULL DEFAULT NULL  AFTER `COL_MY_CITY_INTL`,  ADD `COL_MY_DXCC` INT(6) NULL DEFAULT NULL  AFTER `COL_MY_COUNTRY_INTL`,  ADD `COL_MY_FISTS` INT(10) NULL DEFAULT NULL  AFTER `COL_MY_DXCC`,  ADD `COL_MY_IOTA_ISLAND_ID` VARCHAR(10) NULL DEFAULT NULL  AFTER `COL_MY_FISTS`,  ADD `COL_MY_NAME_INTL` VARCHAR(255) NULL DEFAULT NULL  AFTER `COL_MY_IOTA_ISLAND_ID`,  ADD `COL_MY_POSTCODE_INTL` VARCHAR(255) NULL DEFAULT NULL  AFTER `COL_MY_NAME_INTL`,  ADD `COL_MY_RIG_INTL` VARCHAR(255) NULL DEFAULT NULL  AFTER `COL_MY_POSTCODE_INTL`,  ADD `COL_MY_SIG_INTL` VARCHAR(255) NULL DEFAULT NULL  AFTER `COL_MY_RIG_INTL`,  ADD `COL_MY_SIG_INFO_INTL` VARCHAR(255) NULL DEFAULT NULL  AFTER `COL_MY_SIG_INTL`,  ADD `COL_MY_SOTA_REF` VARCHAR(50) NULL DEFAULT NULL  AFTER `COL_MY_SIG_INFO_INTL`,  ADD `COL_MY_STREET_INTL` VARCHAR(255) NULL DEFAULT NULL  AFTER `COL_MY_SOTA_REF`,  ADD `COL_MY_USACA_COUNTIES` VARCHAR(255) NULL DEFAULT NULL  AFTER `COL_MY_STREET_INTL`,  ADD `COL_MY_VUCC_GRIDS` VARCHAR(50) NULL DEFAULT NULL  AFTER `COL_MY_USACA_COUNTIES`,  ADD `COL_NAME_INTL` VARCHAR(255) NULL DEFAULT NULL  AFTER `COL_MY_VUCC_GRIDS`,  ADD `COL_NOTES_INTL` LONGTEXT NULL DEFAULT NULL  AFTER `COL_NAME_INTL`,  ADD `COL_QRZCOM_QSO_UPLOAD_DATE` DATETIME NULL DEFAULT NULL  AFTER `COL_NOTES_INTL`,  ADD `COL_QRZCOM_QSO_UPLOAD_STATUS` VARCHAR(10) NULL DEFAULT NULL  AFTER `COL_QRZCOM_QSO_UPLOAD_DATE`,  ADD `COL_QSO_DATE` DATE NULL DEFAULT NULL  AFTER `COL_QRZCOM_QSO_UPLOAD_STATUS`,  ADD `COL_QSO_DATE_OFF` DATE NULL DEFAULT NULL  AFTER `COL_QSO_DATE`,  ADD `COL_QTH_INTL` VARCHAR(255) NULL DEFAULT NULL  AFTER `COL_QSO_DATE_OFF`,  ADD `COL_REGION` VARCHAR(25) NULL DEFAULT NULL  AFTER `COL_QTH_INTL`,  ADD `COL_RIG_INTL` VARCHAR(255) NULL DEFAULT NULL  AFTER `COL_REGION`,  ADD `COL_SIG_INTL` VARCHAR(255) NULL DEFAULT NULL  AFTER `COL_RIG_INTL`,  ADD `COL_SIG_INFO_INTL` VARCHAR(255) NULL DEFAULT NULL  AFTER `COL_SIG_INTL`;

ALTER TABLE `TABLE_HRD_CONTACTS_V01` ADD `COL_SOTA_REF` VARCHAR(30) NULL DEFAULT NULL AFTER `COL_SIG_INFO_INTL`, ADD `COL_SUBMODE` VARCHAR(25) NULL DEFAULT NULL AFTER `COL_SOTA_REF`, ADD `COL_UKSMG` VARCHAR(64) NULL DEFAULT NULL AFTER `COL_SUBMODE`, ADD `COL_USACA_COUNTIES` VARCHAR(255) NULL DEFAULT NULL AFTER `COL_UKSMG`, ADD `COL_VUCC_GRIDS` VARCHAR(255) NULL DEFAULT NULL AFTER `COL_USACA_COUNTIES`;
```

Manawyrm, 2019-02-28:
Sure, should I do it, or do you want to create the migration? 

I'd create a 015-logbook.php for this, 
Use individual alter table commands for each field, not one (or two) single ones. 
If a column already exists, that query will fail with a duplicate column error, but that's not bad. 
If it doesn't, it'll get created... With a multiquery like that, it's stop if any column already exists... 

Make sure to set that debug -> false thing from the 014-migration, in order to just skip on errors. 
and get the TABLE_HRD_ name from the config, not just hardcoded. 

That should work very well, then and not cause any trouble for existing users :)

closed 2019-05-31 by magicbug