magicbug/Cloudlog #268: Database error - missing table

g7kse, 2019-03-06:
**Describe the bug**
On  fresh install and after an intial update. I went to import the old adif and get a database error

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Admin > Update Tools > Update DXCC
2. Click on Update
3. Downloading file (seems to take an age)
4. Check in tables to see if dxcc table exists (it does, as does a single dxccexceptions)
5. Go to Admin > adif import
6. Browse and upload

**Expected behavior**
Adif to upload as normal, error as screenshot below. Note also the layout of the screen changes too

**Screenshots**
[image]

Layout change from 
[image]

To
[image]

**Desktop (please complete the following information):**
 - OS: RPi 3 LAMP (Mariadb). DietPi OS (previously ok)
 - Browser Chrome
 - Version 65.0.2 (64-bit)

**Additional context**
Tables are as follows:
[image]

I think I had a previous error and created the table manullay but can't remember what I did and couldn't see a similar error in issues.

magicbug, 2019-03-06:
This issue is now resolved in the code base.

g7kse, 2019-03-06:
I think it is still missing the dxcc_prefixes table

g7kse, 2019-03-06:
Closed after update to code base

closed 2019-03-06 by g7kse