magicbug/Cloudlog #270: Database errors adding QSOs on new install

mich181189, 2019-05-06:
I installed from git master and I'm getting a few errors trying to log anything. Initially I tried importing an ADIF and I got this:

> Incorrect double value: '' for column `m0isrlog`.`TABLE_HRD_CONTACTS_V01`.`COL_A_INDEX` at row 1
> 
> INSERT INTO `TABLE_HRD_CONTACTS_V01` (`COL_A_INDEX`, `COL_ADDRESS`, `COL_ADDRESS_INTL`, `COL_AGE`, `COL_ANT_AZ`, `COL_ANT_EL`, `COL_ANT_PATH`, `COL_ARRL_SECT`, `COL_AWARD_GRANTED`, `COL_AWARD_SUMMITED`, `COL_BAND`, `COL_BAND_RX`, `COL_BIOGRAPHY`, `COL_CALL`, `COL_CHECK`, `COL_CLASS`, `COL_CLUBLOG_QSO_UPLOAD_DATE`, `COL_CLUBLOG_QSO_UPLOAD_STATUS`, `COL_CNTY`, `COL_COMMENT`, `COL_COMMENT_INTL`, `COL_CONT`, `COL_CONTACTED_OP`, `COL_CONTEST_ID`, `COL_COUNTRY`, `COL_COUNTRY_INTL`, `COL_CQZ`, `COL_CREDIT_GRANTED`, `COL_CREDIT_SUBMITTED`, `COL_DARC_DOK`, `COL_DISTANCE`, `COL_DXCC`, `COL_EMAIL`, `COL_EQ_CALL`, `COL_EQSL_QSL_RCVD`, `COL_EQSL_QSL_SENT`, `COL_EQSL_QSLRDATE`, `COL_EQSL_QSLSDATE`, `COL_EQSL_STATUS`, `COL_FISTS`, `COL_FISTS_CC`, `COL_FORCE_INIT`, `COL_FREQ`, `COL_FREQ_RX`, `COL_GRIDSQUARE`, `COL_HEADING`, `COL_HRDLOG_QSO_UPLOAD_DATE`, `COL_HRDLOG_QSO_UPLOAD_STATUS`, `COL_IOTA`, `COL_ITUZ`, `COL_K_INDEX`, `COL_LAT`, `COL_LON`, `COL_LOTW_QSL_RCVD`, `COL_LOTW_QSL_SENT`, `COL_LOTW_QSLRDATE`, `COL_LOTW_QSLSDATE`, `COL_LOTW_STATUS`, `COL_MAX_BURSTS`, `COL_MODE`, `COL_MS_SHOWER`, `COL_MY_ANTENNA`, `COL_MY_ANTENNA_INTL`, `COL_MY_CITY`, `COL_MY_CITY_INTL`, `COL_MY_CNTY`, `COL_MY_COUNTRY`, `COL_MY_COUNTRY_INTL`, `COL_MY_CQ_ZONE`, `COL_MY_DXCC`, `COL_MY_FISTS`, `COL_MY_GRIDSQUARE`, `COL_MY_IOTA`, `COL_MY_IOTA_ISLAND_ID`, `COL_MY_ITU_ZONE`, `COL_MY_LAT`, `COL_MY_LON`, `COL_MY_NAME`, `COL_MY_NAME_INTL`, `COL_MY_POSTAL_CODE`, `COL_MY_POSTCODE_INTL`, `COL_MY_RIG`, `COL_MY_RIG_INTL`, `COL_MY_SIG`, `COL_MY_SIG_INFO`, `COL_MY_SIG_INFO_INTL`, `COL_MY_SIG_INTL`, `COL_MY_SOTA_REF`, `COL_MY_STATE`, `COL_MY_STREET`, `COL_MY_STREET_INTL`, `COL_MY_USACA_COUNTIES`, `COL_MY_VUCC_GRIDS`, `COL_NAME`, `COL_NAME_INTL`, `COL_NOTES`, `COL_NOTES_INTL`, `COL_NR_BURSTS`, `COL_NR_PINGS`, `COL_OPERATOR`, `COL_OWNER_CALLSIGN`, `COL_PFX`, `COL_PRECEDENCE`, `COL_PROP_MODE`, `COL_PUBLIC_KEY`, `COL_QRZCOM_QSO_UPLOAD_DATE`, `COL_QRZCOM_QSO_UPLOAD_STATUS`, `COL_QSL_RCVD`, `COL_QSL_RCVD_VIA`, `COL_QSL_SENT`, `COL_QSL_SENT_VIA`, `COL_QSL_VIA`, `COL_QSLMSG`, `COL_QSLRDATE`, `COL_QSLSDATE`, `COL_QSO_COMPLETE`, `COL_QSO_DATE`, `COL_QSO_DATE_OFF`, `COL_QSO_RANDOM`, `COL_QTH`, `COL_QTH_INTL`, `COL_REGION`, `COL_RIG`, `COL_RIG_INTL`, `COL_RST_RCVD`, `COL_RST_SENT`, `COL_RX_PWR`, `COL_SAT_MODE`, `COL_SAT_NAME`, `COL_SFI`, `COL_SIG`, `COL_SIG_INFO`, `COL_SIG_INFO_INTL`, `COL_SIG_INTL`, `COL_SILENT_KEY`, `COL_SKCC`, `COL_SOTA_REF`, `COL_SRX`, `COL_SRX_STRING`, `COL_STATE`, `COL_STATION_CALLSIGN`, `COL_STX`, `COL_STX_STRING`, `COL_SUBMODE`, `COL_SWL`, `COL_TEN_TEN`, `COL_TIME_ON`, `COL_TIME_OFF`, `COL_TX_PWR`, `COL_UKSMG`, `COL_USACA_COUNTIES`, `COL_VUCC_GRIDS`, `COL_WEB`) VALUES ('', '', '', '', '', '', '', '', '', '', '40M', NULL, '', 'E74BYZ', '', '', '', '', '', '', '', '', '', '', '', '', NULL, '', '', '', '', 'Not Found', '', '', '', '', '', '', '', '', '', '', '70746000', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', 'FT8', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', 'N', '', '', '', '', '', '', '', '', '20181217', '', '', '', '', '', '', '', '-24', '-03', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '', '2018-12-17 12:58', '2018-12-17 12:59', '', '', '', '', '')
> 
> Filename: models/Logbook_model.php
>
> Line Number: 223

I then tried to log a single test QSO and got this:

> Incorrect integer value: '' for column `m0isrlog`.`TABLE_HRD_CONTACTS_V01`.`COL_CQZ` at row 1
>
> INSERT INTO `TABLE_HRD_CONTACTS_V01` (`COL_TIME_ON`, `COL_TIME_OFF`, `COL_CALL`, `COL_BAND`, `COL_FREQ`, `COL_MODE`, `COL_RST_RCVD`, `COL_RST_SENT`, `COL_NAME`, `COL_COMMENT`, `COL_SAT_NAME`, `COL_SAT_MODE`, `COL_COUNTRY`, `COL_MY_RIG`, `COL_QSLSDATE`, `COL_QSLRDATE`, `COL_QSL_SENT`, `COL_QSL_RCVD`, `COL_QSL_SENT_VIA`, `COL_QSL_RCVD_VIA`, `COL_QSL_VIA`, `COL_OPERATOR`, `COL_QTH`, `COL_PROP_MODE`, `COL_IOTA`, `COL_MY_GRIDSQUARE`, `COL_DISTANCE`, `COL_FREQ_RX`, `COL_BAND_RX`, `COL_ANT_AZ`, `COL_ANT_EL`, `COL_A_INDEX`, `COL_AGE`, `COL_TEN_TEN`, `COL_TX_PWR`, `COL_STX`, `COL_SRX`, `COL_NR_BURSTS`, `COL_NR_PINGS`, `COL_MAX_BURSTS`, `COL_K_INDEX`, `COL_SFI`, `COL_RX_PWR`, `COL_LAT`, `COL_LON`, `COL_DXCC`, `COL_CQZ`, `COL_GRIDSQUARE`) VALUES ('2019-05-06 23:17', '2019-05-06 23:17', 'M1ABC', '20m', 14200000, 'SSB', '59', '59', 'name', 'comment', '', '', '', NULL, '2019-05-06', '2019-05-06', 'N', NULL, '', NULL, '', 'M0ISR', 'location', '', '', 'IO91XL', '0', 0, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, NULL, '', '', 'LOCATOR')

It looks like the ADIF importer is inserting empty strings when it should be inserting nulls. I tried to fix these cases, however I then got 

> Data too long for column 'COL_DXCC' at row 1

magicbug, 2019-05-11:
Have you got an Adif file I can see?

magicbug, 2019-05-11:
Also what's the server setup PHP & mysql versions

mich181189, 2019-05-11:
> $ php -v
> PHP 7.2.17 (cli) (built: Apr  2 2019 15:29:06) ( NTS )
> Copyright (c) 1997-2018 The PHP Group
> Zend Engine v3.2.0, Copyright (c) 1998-2018 Zend Technologies

> $ mariadb_config --version
> 10.3.6

As for an ADIF file, I used one from Clublog. I also had problems with inserting via the cloudlog API though (CQZ column empty - again, it could be expecting NULL rather than '')

TABLE_HRD_CONTACTS_V01 is using the MyISAM engine.

magicbug, 2019-05-11:
Other thing is under "admin" you have run the update country files?

mich181189, 2019-05-11:
Ok so "update country files" did help although I got a lot of errors on import, it seems to have done roughly the right thing.

mich181189, 2019-05-11:
Ah right, not quite true - I still had some of my attempted fixes in Just tried the import without my fixes:

> Incorrect double value: '' for column `m0isrlog`.`TABLE_HRD_CONTACTS_V01`.`COL_A_INDEX` at row 1

mich181189, 2019-05-11:
Aha so the live log error (cqz not a valid number) was actually because I was missing the php-json package

The ADIF import I'm still looking at. It's trying to insert "Not Found" which is "too long"

mich181189, 2019-05-11:
So... One problem was that the update directory was read only to the webserver. Another was that I was missing the php-xml package, so the country update was (mostly silently) failing.

Incoming PR for the DB model fixes.

magicbug, 2019-05-12:
Thanks will add a note about PHP modules & check the pr in the morning

mich181189, 2019-05-13:
I found a few more in the DXCC importer, I started fixing them but then I hit this problem I haven't had chance to fix yet:

> Data too long for column 'call' at row 1
> 
> INSERT INTO `dxcc_exceptions` (`record`, `call`, `entity`, `adif`, `cqz`, `cont`, `long`, `lat`, `start`, `end`) VALUES (40844, 'VE3CUS/VP2K', 'ANGUILLA', 12, 8, 'NA', -63, 18.3, '1968-03-13 00:00:00', '1968-03-22 23:59:59')
> 
> Filename: controllers/Update.php

magicbug, 2019-05-19:
Thanks for the PRs this has been merged, need to rework the update screen so that it isn't done in the background but instead visible so that errors can be caught.

closed 2019-05-22 by magicbug