magicbug/Cloudlog #276: DXCC_Entities table will be empty after update out of program

pa2eon, 2019-05-30:
Peter,

When I go to the program option 'Admin - Update Country File' the table list in dxcc_entities will be empty after that !

So this will give the problem, when you want to make a new location profile.
OR ..
The program MUST have a fresh dxx list / or file somewhere .. where it is not find.
The install folder I had renamed to install_old .. is that needed or is that wrong ? 

vy '73 - PA2EON

magicbug, 2019-05-31:
If this is a fresh install follow the guide in the wiki, the country files are downloaded from clublog then imported into Cloudlog

closed 2019-05-31 by magicbug

pa2eon, 2019-06-01:
Yes I did .. the ./updates folder you have to make read/write ( 777 ).
The *.gz file is untar to the file 'cty.xml'
But no data is going to the sql table 'dxcc_entities' , may be the structure is not good ?