magicbug/Cloudlog #278: Import dxcc entities will not fill database

pa2eon, 2019-06-01:
The update.php in /application/controllers/update.php have to fuill in the database tables:
dxcc_entities / dxcc_exceptions / dxcc_prefixes / dxccexceptions
But thsi is not done by the update button from the program.

You have to make the folder / updates/  R/W ( 777 ).
The cty.xml file from clublog is downloaden and placed in this folder, but the 'engine' did not update the tables .. where to look for .. no errors will be given.

vy '73, PA2EON

magicbug, 2019-06-01:
Since you changed your config for MySQL did this resolve the above issue?

pa2eon, 2019-06-01:
Sorry no .. the tables will be made empty first ( that should be fine, when you follow the source code) but after that no new data is filled in . it seems that the data is lost before it have to fill the tables.

pa2eon, 2019-06-03:
After the correct 'plugin' for PHP7.x and Apache on Ubuntu 16.04.x everthing works fine now:
These are the needed updates:

sudo apt-get install zlib1g-dev libxml2-dev
sudo cpan install XML::LibXML
!
sudo apt-get install php7.2-zip
!
apt-get install php7.2-gd 
!
sudo nano /etc/mysql/conf.d/disable_strict_mode.cnf (make this new file ! )
  Add the next lines:
  [mysqld]
sql_mode=IGNORE_SPACE,NO_ZERO_IN_DATE,NO_ZERO_DATE,ERROR_FOR_DIVISION_BY_ZERO,NO_AUTO_CREATE_USER,NO_ENGINE_SUBSTITUTION
Save the file.

These files are needed when you use a Ubuntu Server 16.04 OS with PHP7.2.x and Apache.

Regards, PA2EON

closed 2019-06-03 by magicbug