magicbug/Cloudlog #285: COL_QSL_VIA is not in edit QSO dialog

dg9vh, 2019-06-12:
The QSL-Manager-Field (QSL via) is missing in edit-qso-dialog. It is existing in new QSO-dialog.

73 de Kim
DG9VH

dg9vh, 2019-06-12:
This issue is based on the absence of /css and /js-folders referenced in application/views/layout/mini_header.php

Here are some lines like 
<link rel="stylesheet" href="<?php echo base_url();?>css/reset.css" type="text/css" />
and similar where those files / folders not existing within the project... therefore the edit-window shows up unstyled and non-functional.

I think it is related to https://github.com/magicbug/Cloudlog/issues/281 as you are migration to new version of bootstrap etc...

magicbug, 2019-06-13:
Yes known and due to ongoing UI upgrades, the edit window functions just completely unstyled

magicbug, 2019-06-13:
This is now shown in the edit dialogue box

closed 2019-06-13 by magicbug