magicbug/Cloudlog #287: Adding Description-Label to API-Key

dg9vh, 2019-06-14:
It would be useful to add a description field to the API-Keys so one can put a label to a key to identify it's usage and make revoking/deleting easier, if necessary.

Due to security reasons one would create different API-Keys for each new application connected to the API to have the possibility to deny access later for a special service by revoking the single key and let only one application die :-)

magicbug, 2019-06-14:
Yes, I have them for separate apps was meaning to add a description its a good suggestion!

closed 2019-07-09 by magicbug

dg9vh, 2019-07-10:
Working nice and perfect :-)

dg9vh, 2019-10-02:
@magicbug this issue is still listed on https://github.com/magicbug/Cloudlog/projects/5#column-5795525 as "in progress"... I think, you should mark it as "Done"...