magicbug/Cloudlog #288: Wrong Frequency on ADIF-Import

dg9vh, 2019-06-14:
Hi,

when importing following ADIF-file, the Frequency is imported incorrect for example for first entry as 70.41500 that is mapped as 4m QSO.

```
File: export.adi
<ADIF_VER:5>2.2.7
<PROGRAMID:6>fldigi
<PROGRAMVERSION:6>4.1.03
<EOH>
<FREQ:8>7.041500<CALL:6>GI4SZW<MODE:3>PSK<SUBMODE:5>PSK31<NAME:6>Seamus<QSO_DATE:8>20190603<QSO_DATE_OFF:8>20190603<TIME_ON:6>190100<RST_RCVD:3>599<RST_SENT:3>599<EOR>
<FREQ:8>7.044980<CALL:7>GB19CGW<MODE:4>RTTY<NAME:4>Rsgb<QSO_DATE:8>20190603<QSO_DATE_OFF:8>20190603<TIME_ON:6>192800<RST_RCVD:3>599<RST_SENT:3>599<EOR>
<FREQ:8>7.040677<CALL:6>LY2BKQ<MODE:3>PSK<SUBMODE:5>PSK31<NAME:8>Anatolij<QSO_DATE:8>20190610<QSO_DATE_OFF:8>20190610<TIME_ON:6>184500<RST_RCVD:3>599<RST_SENT:3>599<EOR>
<FREQ:8>7.041223<CALL:8>SM5LIB/P<MODE:3>PSK<SUBMODE:5>PSK31<NAME:4>Igge<QSO_DATE:8>20190610<QSO_DATE_OFF:8>20190610<TIME_ON:6>185500<RST_RCVD:3>599<RST_SENT:3>599<EOR>
```

How could this be solved? Should band-data be exported with to get frequency correct?

Kim
DG9VH

magicbug, 2019-06-14:
Seems we have an issue with the import of the frequency for example WSJT-X 40m - freq is stored as 7074987 so 7.074.987 which cloudlog handles fine but coughing on the FLdigi

This is handled by https://github.com/magicbug/Cloudlog/blob/master/application/models/Logbook_model.php in the import() function I'll take a look tomorrow and try make some adjustments

Peter.

magicbug, 2019-06-14:
Converting frequency to the band is handled by https://github.com/magicbug/Cloudlog/blob/master/application/libraries/Frequency.php

dg9vh, 2019-06-14:
Idea: getting frequency as string and remove all dots to get a normalized format... or, if band data is given in adif convert to correct format based on this knowledge.

magicbug, 2019-06-14:
Agree that does seem the best solution to strip out the dots!

magicbug, 2019-06-14:
I did actually think we striped out the dots, but could be wrong, need to check it all tomorrow when I'm less half asleep

magicbug, 2019-06-14:
Out of interest inside the database whats the frequency stored?

magicbug, 2019-06-15:
I'll look into this in the morning, the frequency should be stripped of periods on import, it could be that it's not happening however it seems to with other Adif files.

I'll report back soon

magicbug, 2019-06-29:
Hi @dg9vh, I believe this issue is now resolved if you try https://github.com/magicbug/Cloudlog/commit/de9cd471f57a44a89033827ebf959d003e9b96a0

dg9vh, 2019-06-29:
Would try it, if having some PSK-QSOs again :-) tnx so far. I let it open so long, ok?

magicbug, 2019-06-29:
If i feel crazy enough, i might have a little look on PSK later or in the morning

dg9vh, 2019-06-29:
well, I don't know, if I got time this evening :-) got a hard day in the garden today, you know :-)

dg9vh, 2019-06-29:
So, now I imported two(!) QSOs from FLDigi.

Following file:
File: export.adi
<ADIF_VER:5>2.2.7
<PROGRAMID:6>fldigi
<PROGRAMVERSION:6>4.1.04
<EOH>
<FREQ:8>7.041103<CALL:6>R39PGD<MODE:3>PSK<SUBMODE:5>PSK63<QSO_DATE:8>20190629<QSO_DATE_OFF:8>20190629<TIME_OFF:6>194724<TIME_ON:6>194500<RST_RCVD:3>599<RST_SENT:3>599<BAND:3>40m<COUNTRY:15>European Russia<CQZ:2>16<STX:3>000<OPERATOR:5>DG9VH<STATION_CALLSIGN:5>DG9VH<MY_GRIDSQUARE:6>JN39kf<MY_CITY:11>Voelklingen<EOR>
<FREQ:8>7.041247<CALL:7>G4EBY/P<MODE:3>PSK<SUBMODE:5>PSK31<NAME:7>Gregory<QSO_DATE:8>20190629<QSO_DATE_OFF:8>20190629<TIME_OFF:6>195351<TIME_ON:6>195000<QTH:12>High Wycombe<RST_RCVD:3>599<RST_SENT:3>599<GRIDSQUARE:6>IO03cc<BAND:3>40m<COUNTRY:7>England<CQZ:2>14<STX:3>000<OPERATOR:5>DG9VH<STATION_CALLSIGN:5>DG9VH<MY_GRIDSQUARE:6>JN39kf<MY_CITY:11>Voelklingen<EOR>

and *drums* it works as expected!

Good work :-)

73 de Kim
DG9VH

closed 2019-06-29 by dg9vh