magicbug/Cloudlog #289: Fresh install, users table missing

M0LTE, 2019-06-14:
Freshly installed Ubuntu 18.04 Server LTS

- Install apache2, mysql, php
- git clone repo
- create mysql database, user
- load /install in a browser
- fill in the wizard
- get success message

browse to root of site, get the following:

```
A Database Error Occurred
Error Number: 1146
Table 'cloudlog.users' doesn't exist
SELECT * FROM `users` WHERE `user_id` IS NULL
Filename: models/User_model.php
Line Number: 34
```

Sure enough:
```
mysql> show databases;
+--------------------+
| Database           |
+--------------------+
| information_schema |
| cloudlog           |
| mysql              |
| performance_schema |
| sys                |
+--------------------+
5 rows in set (0.00 sec)

mysql> use cloudlog
Reading table information for completion of table and column names
You can turn off this feature to get a quicker startup with -A

Database changed
mysql> show tables;
+--------------------+
| Tables_in_cloudlog |
+--------------------+
| api                |
| cat                |
| config             |
| contest_template   |
| contests           |
| dxcc               |
| dxccexceptions     |
| migrations         |
+--------------------+
8 rows in set (0.00 sec)
```

magicbug, 2019-06-15:
If you look in /install/assets there's an SQL file, however I know it's there's as I've done two installs today.

This hints at another issue, what PHP version?

magicbug, 2019-06-15:
@M0LTE other options to import the SQL dump inside /install/assets manually to see if there's any unusual errors

M0LTE, 2019-06-15:
```
# php --version
PHP 7.2.19-0ubuntu0.18.04.1 (cli) (built: Jun  4 2019 14:48:12) ( NTS )
```

will look at other stuff now. Cheers Pete.

magicbug, 2019-06-15:
PHP version is :+1: that's what I have only production box.

This feels like it's going to end up being mysql related

M0LTE, 2019-06-15:
```
ERROR 1292 (22007) at line 3381: Incorrect datetime value: '0000-00-00 00:00:00' for column 'start' at row 1
```

That line is:
```
INSERT INTO `dxccexceptions` VALUES ('1', 'HF0QF', 'SOUTH SHETLAND', '13', '73', 'SA', '-58.3', '-62', '0000-00-00 00:00:00', '0000-00-00 00:00:00');
```

M0LTE, 2019-06-15:
```
mysql> SHOW VARIABLES LIKE 'sql_mode' ;
+---------------+-------------------------------------------------------------------------------------------------------------------------------------------+
| Variable_name | Value                                                                                                                                     |
+---------------+-------------------------------------------------------------------------------------------------------------------------------------------+
| sql_mode      | ONLY_FULL_GROUP_BY,STRICT_TRANS_TABLES,NO_ZERO_IN_DATE,NO_ZERO_DATE,ERROR_FOR_DIVISION_BY_ZERO,NO_AUTO_CREATE_USER,NO_ENGINE_SUBSTITUTION |
+---------------+-------------------------------------------------------------------------------------------------------------------------------------------+
1 row in set (0.01 sec)
```

NO_ZERO_DATE is the cause

magicbug, 2019-06-15:
Looks like this has changed in mysql and old cruft from yester year has caught up.

Temp solution is

SET SQL_MODE='ALLOW_INVALID_DATES';

However I think I can delete those bits from SQL long term, another weekend job

M0LTE, 2019-06-15:
Stuck that at the top of the SQL file, seemed to work ok, I'm now installed and logged in.

Cheers Pete.

M0LTE, 2019-06-15:
So I guess there’s a couple of improvements possible:

1. Fix the SQL to work with modern MySQL out of the box
2. Fix the thing that runs the script to stop and report on errors (it reported success to the GUI to me)

Top marks for responsiveness last night, thank you 👍🏼

magicbug, 2019-06-15:
Insomnia is good for doing support, I can remove the offending parts from the SQL which should fix it

magicbug, 2019-06-18:
This should now be fixed

closed 2019-06-18 by magicbug