magicbug/Cloudlog #291: Feature-Request: QSL-Received Button beneath "Edit"-Button in Logbook-View

dg9vh, 2019-06-15:
For faster paper-QSL-card recording, a button called "QSL received" with the functionality described below would be of effort:

Functionality:
When the button is clicked, set COL_QSL_RCVD = 'Y' and COL_QSLRDATE = now().

This speeds up mass-registration of received QSL-cards after a club meeting. If you want to register actually, you need to do 3 clicks after searching the callsign and identifying the qso in the list. The new button speeds up by 1/3 of time at least.

If you agree, I could try to realize it in a separate branch on a fork of your code and you can take a look at it before merging it into the master-branch.

73
Kim
DG9VH

dg9vh, 2019-06-15:
Just made a PR for this. here on my side it's working.

magicbug, 2019-06-15:
Thanks all, always nice when i can just merge a PR :)

dg9vh, 2019-06-15:
I know - and with my few skills, I could bring some ideas into your vy gd project :-) So a classical win-win-situation :-)

magicbug, 2019-06-16:
OK just had a morning with the new button, it's annoying me a bit, but not sure the best way to address it, I do wonder whether the button could be removed and the arrow for incoming QSL could be the link instead making it a bit more subtle, the same could be added for sent too

magicbug, 2019-06-16:
Thinking further I'm wondering whether this could be a Cog icon on the logbook display than a drop down which could support more options.

magicbug, 2019-06-16:
Made some adjustments as the PR was missing a couple of files and it didn't work, Edit/QSL Received Marking is now a dropdown cog which will allow for more features.

I need to add similar to the search

closed 2019-06-16 by magicbug

dg9vh, 2019-06-16:
This is a much better way to do it :-) and also a speeding up at qsl-receiving :-)