magicbug/Cloudlog #293: API for adding QSOs

M0LTE, 2019-06-15:
**Is your feature request related to a problem? Please describe.**
There doesn't seem to be an API for pushing in QSOs from external programs

**Describe the solution you'd like**
I would like to be able to POST some JSON a bit like this to https://my-cloudlog/index.php/api/qso (with API key)
```
{
  "time": "2019-06-15T22:06.05Z",
  "freq": 10136750,
  "mode": "FT8",
  "theircall":"2E1EPQ",
  "theirgrid": "JN02",
  "recvrpt": "-10",
  "sentrpt": "+04",
  "mygrid": "IO91lk",
  "operator": "M0LTE"
}
```
I plan to write a tool to watch wsjtx_log.adi and post to that API to enable auto-logging - though the same mechanism could be used by other tools.

**Describe alternatives you've considered**
There isn't really an alternative

**Additional context**
n/a

magicbug, 2019-06-16:
Yes, something I'm currently adding per a request from M1BXF & M0VFC I was going to just do it so you could submit ADIF string, but I can do it so it supports both.

M0LTE, 2019-06-16:
Either is good :)

Started hacking something around last night but figured you’d probably want
to handle something so core yourself, so I stopped.

Cheers
Tom

magicbug, 2019-06-16:
If you fancy adding a json function for QSO then feel free Tom, not touched it yet.

magicbug, 2019-06-17:
Hi Tom,

It is now possible to do it with json post to the following URL `/index.php/api/qso`

**Example**

`{"key":"YOUR_API_KEY", "type":"adif",
"string":"<call:5>N9EAT<band:4>70cm<mode:3>SSB<freq:10>432.166976<qso_date:8>20190616<time_on:6>170600<time_off:6>170600<rst_rcvd:2>59<rst_sent:2>55<qsl_rcvd:1>N<qsl_sent:1>N<country:24>United States Of America<gridsquare:4>EN42<sat_mode:3>U/V<sat_name:4>AO-7<prop_mode:3>SAT<name:5>Marty<eor>"
}`

It can actually handle multiple lines, I do need to add some error handing notifications but it does work.

M0LTE, 2019-06-17:
Good stuff Peter :) ADIF inside JSON ;-) works for my needs.

Is that example the complete set of ADIF fields supported? If I send more
will they be correctly silently ignored? Or do I need to clean up lines
before sending them in?

Have you made use of HTTP status codes for success/failure?

Cheers
Tom

magicbug, 2019-06-17:
It supports all ADIF fields, if you have them Cloudlog will chew all the data into the table.

Its a complete bodge at the moment, I will go back to it and add proper error handling, I'll also add the ability to send QSOs as per your request rather than just ADIF this just meets the need of a few requests I've had.

M0LTE, 2019-06-17:
Look forward to trying it.

Cheers Peter.

magicbug, 2019-06-17:
401 for failed due to api
201 if records created successfully 

are currently set

magicbug, 2019-06-17:
Now just need someone to build an app that can monitor local folders for ADIF files and send qsos to Cloudlog automatically saving uploading! :)

M0LTE, 2019-06-17:
I’m on it!!

closed 2019-06-18 by magicbug