magicbug/Cloudlog #294: FR: Clublog-Upload/Export

dg9vh, 2019-06-16:
I don't know, if Clublog is providing an API for this but you also can deliver log-entries via E-mail, so it could be done with an additional field in the logbook-table that marks a log-line as "delivered to clublog" after exporting it and sending it to Clublog.

This is a "nice to have"-feature - actually the ADIF-export is working for this but one has to upload it by hand (or by email) to Clublog.

73 de Kim
DG9VH

magicbug, 2019-06-16:
Its been on my radar for a while, Clublog is already used for the dxcc / exception information within Cloudlog. I need to look at the best way to integrate some real time qso posting, really its probably better to mark it as a task and its handled by another script

dg9vh, 2019-06-16:
Yes, I agree - doing it automatically by logging a QSO could cause some problems when logging "by accident"... so triggering the export by the user would be the best way.

magicbug, 2019-06-16:
Okay, think the best way to do this create a controller called **clublog** then  process

- sync - which sends qsos to Clublog and also downloads from clublog which marks the COL_CLUBLOG_QSO_UPLOAD_DATE and COL_CLUBLOG_QSO_UPLOAD_STATUS fields

then use the config file to store the information Cloudlog needs to talk with Clublog.

Thoughts? @dg9vh @Manawyrm

dg9vh, 2019-06-16:
What exactly you would fetch from clublog? Syncing up to clublog ok in this way, but what to sync downwards? Syncing-results like changes of DXCC or similar?

magicbug, 2019-06-19:
Okay so I have added a new controller with a function `clublog/upload/<username>` if you have added your Clublog details to your user profile and provide the username with your username you can then upload to Clublog it will then mark the QSOs as sent.

dg9vh, 2019-06-19:
When calling the URL with my callsign as registered as callsign at clublog, I get following message without nothing else:

Nothing awaiting upload to clublog

magicbug, 2019-06-19:
Yes seems issue with anything beyond my local machine!

dg9vh, 2019-06-19:
ok, seems correct:
waiting a few minutes and calling again:
Club Log has already processed this exact file in the past (checksum duplicate)
A PHP Error was encountered
Severity: Warning

Message: strpos() expects parameter 1 to be string, resource given

Filename: controllers/Clublog.php

Line Number: 72

QSOs uploaded and Logbook QSOs marked as sent to Clublog

Only thing now: Warning :-)

magicbug, 2019-06-19:
Okay, just committed code thats working locally for me, however its not working on production not even getting a warning

magicbug, 2019-06-19:
Think i found it oops

dg9vh, 2019-06-19:
ok, waiting for new code :-)

dg9vh, 2019-06-19:
With actual code getting following:
2.2 Cloudlog Version 1.7 
Fatal error: Allowed memory size of 268435456 bytes exhausted (tried to allocate 20480 bytes) in /home/huebelon/public_html/dg9vh.de/cloudlog/system/database/drivers/mysqli/mysqli_result.php on line 237

Fatal error: Allowed memory size of 268435456 bytes exhausted (tried to allocate 32768 bytes) in /home/huebelon/public_html/dg9vh.de/cloudlog/system/core/Exceptions.php on line 1

magicbug, 2019-06-19:
OK, stupid me error, forgot to add the directory array from the config.php to the clublog stuff so it was looking for files in the wrong place ... I know idiot.

magicbug, 2019-06-19:
Only thing not fixed now is marking QSOs as sent to clublog

dg9vh, 2019-06-19:
[image]

Seems to work now :-)

magicbug, 2019-06-19:
Just need to fix marking QSOs as sent to Clublog then I'll be happy.
Hopefully have some time tomorrow evening!

magicbug, 2019-06-20:
Should be resolved now with any luck, mine are going up ok

dg9vh, 2019-06-20:
It's working after "clear log" at clublog.org... otherwise we simply get an "Error"-Message...

Maybe something, that should be documented later in the wiki?

magicbug, 2019-06-20:
Error message will now display what clublog returns.

closed 2019-06-20 by magicbug