magicbug/Cloudlog #295: FR: Getting QSL-Manager from QRZ.com

dg9vh, 2019-06-16:
As I know, one can provide his login-credentials for qrz.com on the backend. So with this (and with an XML-subscription on qrz.com-side) it would be possible to fetch QSL-Manager-entries (and other data) from qrz.com-database.

I think, this would also be a nice feature (but I also know, that this should be "optional" by config it in preferences, because it depends on an XML-subscription).

We should think this over...

magicbug, 2019-06-16:
QRZ is already supported if you provide login in /application/config/config.php it returns most features like name etc.. I'm not sure if they provide a callsign for QSL Manager need to check that

magicbug, 2019-06-16:
Just looked at the spec for QRZ api at [https://www.qrz.com/XML/current_spec.html](https://www.qrz.com/XML/current_spec.html) it would appear they provide a qslmgr field so that data could be passed across

dg9vh, 2019-06-16:
Actually it seems not to be passed - it is delivered, I know this :-) but actually it is not transported into the log.
I think, this could be easily done :-)

magicbug, 2019-06-16:
Does now :)

dg9vh, 2019-06-16:
Done with https://github.com/dg9vh/Cloudlog/pull/4/commits/53728a851d899749c43cfc5d6c912f13d7ab7fad
You're devil one :-)

dg9vh, 2019-06-16:
Hi,
This function has an issue with entries that where longer than x characters I think. For example: AM70S, here QRZ.com knows following entry:
EA4URE - BUREAU (PREFERED) OR DIRECT

Here nothing is transported. We should consider about filling all of it into the edit-box. Or could this be a problem, because this is also a station worked before and entered EA4URE manually as manager into the field?

magicbug, 2019-06-16:
Yep this is easily solved, seems the database schema (which was actually from HamRadioDeluxe back when it was free) set the field to "60" which is nearly too short, I'll sort a migration file to make this longer

dg9vh, 2019-06-16:
By the way to get this migration-process understanding: The migration-files are started automatically after they exist in the background by calling the website (dashboard)? Or is there anything to do by hand to activate them?

magicbug, 2019-06-16:
Yes they are automatically run in the background don't need todo anything. you create a migration file then set it in the migrate.php config file to the latest number and it runs it.

Okay as per https://github.com/magicbug/Cloudlog/commit/9fb7d82702c276c4453945b098859b72f87f0f45 this should be resolved if you could try it and let me know! the field is now 255.

dg9vh, 2019-06-16:
Yes, looks better now with the long part. Tested it with AM70F, which never worked actually... here correct manager entry is fetched.

But: When trying AM70S (worked before) manager-entry stays empty... Are you fetching infos into the fields from QSOs before? If so, then maybe the manager-info is not fetched from there?

magicbug, 2019-06-16:
If QSO is found already I believe it doesn't hit QRZ it grabs it from last QSO.. which probably means a bug in the code.. will sort that in the morning :D

dg9vh, 2019-06-16:
That's early enough :-)

magicbug, 2019-06-16:
Least we're a step closer, I actually noticed similar with IOTA field although probably don't want that prepopulated....

dg9vh, 2019-06-16:
IOTA makes sense, because one can jump from island to island, but manager would not change so often or if, it would be ok to change it by hand :-)

magicbug, 2019-06-17:
VIA field is populated from logbook too now

dg9vh, 2019-06-17:
Topp!

closed 2019-06-17 by dg9vh