magicbug/Cloudlog #303: Distance in meter (configurable)

dg9vh, 2019-06-24:
Would it be possible (I think somewhere in .../libraries/Qra.php) to calculate the distance based on metric meters than in miles?
We could make it configurable via the known config-way but I have no idea where and how to inject the code to make the transformation. I am a little confused about some "K", "N" and "M" already in the code.

73 de Kim

magicbug, 2019-06-24:
M = miles
K = km (hence * 1609.344)
N = I'm not actually sure why that's there at (miles * 0.8684)

magicbug, 2019-06-24:
Just checked N is Nautical Miles

dg9vh, 2019-06-24:
Was my idea, too. So the transformation is already there, only need to build in the config-item. Woul do this at least this eve.

Kim

magicbug, 2019-06-24:
To point you in the right direction

controller/logbook.php look for searchbearing() and bearing() you'll find both call the bearing function from the library just tag on the unit :)

magicbug, 2019-06-24:
I'll close this as its added and wiki documentation added.

closed 2019-06-24 by magicbug