magicbug/Cloudlog #310: Show warning if selected radio CAT data is X minutes old

magicbug, 2019-06-27:
I'm proposing we add a warning on the QSO page if the selected radios data is older than X minutes to catch any potential issues.

dg9vh, 2019-06-27:
Good idea but what interval are you thinking about? Consider someone's calling CQ or you're having a CW-QSO on 40m, then I would see 30 minutes as a value we can use... or - yeah, new option in cloudlog.php - we make it configurable in the custom-config :-) So everyone using CAT can decide, what interval would be best for his use-cases... good idea?

magicbug, 2019-07-06:
I've started this by adding the option to config/cloudlog.php to default to 5minutes (https://github.com/magicbug/Cloudlog/commit/66455193a8f03e5efc03487ad480454786a898a2) I'll be adding the code to display the message tomorrow.

dg9vh, 2020-04-19:
@magicbug how far is this going on? Is it transported to master or still in development?

magicbug, 2020-04-19:
Barely started if I'm honest keeps getting knocked down my list

dg9vh, 2020-04-19:
Hihi, no problem...

magicbug, 2020-05-12:
[image]

Started working on this today :)

magicbug, 2020-05-12:
This is now implemented and functioning

closed 2020-05-12 by magicbug