magicbug/Cloudlog #312: Edit possible options 'change Station Profile'

pa2eon, 2019-06-30:
1. When edit a QSO, need the possibility to edit the Station Profile name.
No you can not see after from what 'station profile' you had made the QSO.

2. The active edit screen may be bigger in size or change size by yourself.

vy '73 PA2EON

magicbug, 2019-06-30:
Station Profile is set a set of variables that held then put with the QSO record, I will make these text boxes within the edit, quite a few more need to be made available.

magicbug, 2019-06-30:
Made my station profile information available for editing.

closed 2019-06-30 by magicbug

pa2eon, 2019-06-30:
Many thanks, seen the update.
In which source file I can change the viewport settings .. to make the edit active window bigger ?

Eric

magicbug, 2019-06-30:
I'd recommend exploring the /application/views folder there's a lot of code with the potential to break things

pa2eon, 2019-06-30:
Yep I seen .. the best I need is in general.css .. change the parm: .fancybox-content { .. make 200 more ) 
that will do the job .. =;)