magicbug/Cloudlog #313: No check QRZ / HamQTH after git update

pa2eon, 2019-06-30:
Dear Peter,

The latest changes will kill the QRZ/HamQTH update requests.
My previous version (see picture)
[image]
Then after the github update of today:
[image]

So now a great function is lost =;) 

vy '73 Eric / PA2EON

magicbug, 2019-06-30:
Hmm not sure that anything has changed to break it, will look

magicbug, 2019-06-30:
I can confirm that QRZ is certainly working

magicbug, 2019-06-30:
Eric,

Can you confirm that you are seeing this issue both with the QRZ and the HamQTH? as I'm using QRZ and can't see any issue with fields being populated, I'll check HamQTH later.

pa2eon, 2019-06-30:
A requests to HamQTH their will be no information about the station.
A reguest to QRZ.COM the information is collected and shown.

So something for the HamQTH routine is bugged.

vy '73, Eric

magicbug, 2019-06-30:
OK strange, guessing nothing changed within the /application/config/config.php file? 

If you go to `<url-to-cloudlog>/index.php/logbook/json/mm3php` does it return any results or does it error? you might need to edit index.php and set the debug mode to development

magicbug, 2019-06-30:
I've made one code adjustment that if QSL_Manager isn't returned its skipped, not sure whether this might have caused an issue.

However, just tried it again after defining the below items in cloudlog.php and it worked flawlessly 

`
$config['hamqth_username'] = "";
$config['hamqth_password'] = "";
`

closed 2019-06-30 by magicbug