magicbug/Cloudlog #314: After 2nd update error from /migrations/016_lotwusers.php

pa2eon, 2019-06-30:
When I update the software twice from github ( git pull ) I get the error that the record is stil their from the file 016_lotwusers.php

[image]

Also from the file 018_clubloguserfields.php I have to clear all the line's.
To make a work-arround you can add a '#' sign in front of the line.

[image]

Need to update the source code when check if the database fileds are their or not !

vy '73 PA2EON

magicbug, 2019-06-30:
Migrations run when Codeigniter hasn't run them, I've done 6 new installs with the current code and update 15 nightly and its worked flawlessly so I don't really plan on changing them. 

If your having to code change then something has changed within your install and i'm not sure what.

pa2eon, 2019-06-30:
I think that is correct .. the error was in the second 'run' with the git update.
So when the code checks if the tables are their it have to do nothing .. but it will .. so it says something like 'already have the table name .. ed. '

pa2eon, 2019-06-30:
After new github update .. no error anymore ! 
Seems to be fixed ..

closed 2019-06-30 by magicbug