magicbug/Cloudlog #315: Result of direct search in QRZ database after git update

pa2eon, 2019-06-30:
>>
OK strange, guessing nothing changed within the /application/config/config.php file?

If you go to <url-to-cloudlog>/index.php/logbook/json/mm3php does it return any results or does it error? you might need to edit index.php and set the debug mode to development

It will give no error but also no search result !

[image]

The latest souce (git) update will give no errors anymore !

vy '73 Eric

pa2eon, 2019-06-30:
Update the screen to .. no find call information .. and no filled data in fields screen:

[image]

pa2eon, 2019-06-30:
Test the checkup via Hamqth !!!  => The data filled in correct !
Changed to QRZ.com .. the data was NOT filled in (see screendump ) .. weird !!

magicbug, 2019-06-30:
Please make sure you have QRZ information set I'm using an active subscription and it works perfectly.

pa2eon, 2019-06-30:
FOUND THE SOLUTION !!!!  =;)
In QRZ.com you may NOT be use an '#' sign in your password !
Remove it .. and the search result is working !!

magicbug, 2019-06-30:
Odd that # causes an issue but pleased it's resolved

closed 2019-06-30 by magicbug