magicbug/Cloudlog #316: No Map if logged in

dg9vh, 2019-07-01:
**Describe the bug**
After login in, no map is shown on dashboard or logbook-tab.

**To Reproduce**
Steps to reproduce the behavior:
1. Log in
2. Click on Logbook
3. See error

**Expected behavior**
The Map should be shown.

**Error-messages from Code Inspector (Chrome)**

logbook:1219 Uncaught SyntaxError: Unexpected token ;

Corresponding lines in Code:
                var q_lat = ;
        var q_lng = ; 

**Desktop (please complete the following information):**
 - OS: Windows 10 Home
 - Browser: Chrome
 - Version: 75.0.3770.100

magicbug, 2019-07-01:
I'm not seeing that here at all? going to need some more informatoin

magicbug, 2019-07-01:
I'd suggest looking at `/index.php/logbook/qso_map/25/` to see if anything is displayed espescially errors, but I don't believe i broke anything

dg9vh, 2019-07-01:
Hi, 
problem solved. I replaced my QRA-Locator yesterday from JN39KF to JN39KF05 - that was not resolved to longitude and latitude. Removing the last 2 digits, it is solved.

73 de Kim
DG9VH

closed 2019-07-01 by dg9vh