magicbug/Cloudlog #318: Band-slot grid

M0LTE, 2019-07-20:
**Is your feature request related to a problem? Please describe.**
It would be really nice to be able to see a summary of number of QSOs per country per hand.

**Describe the solution you'd like**
A grid. Along the top, bands. Along the left side, countries. In each cell, count of QSOs in that slot.
Only show band columns with QSOs.
Check a checkbox to show countries with zero contacts in any band - needed slots

**Describe alternatives you've considered**
Considered writing the DB query myself ;-)

**Additional context**
none

M0LTE, 2019-07-20:
Could link to this page from Countries Breakdown... Worked and Needed on the homepage

M0LTE, 2019-07-20:
It would probably live under the Analytics tab

magicbug, 2019-07-21:
Is this not covered by Awards - DXCC?

M0LTE, 2019-07-21:
Oops, totally missed this page. Pretty much, yes.

Would be nice to be able to show missing countries, and hide bands with no QSOs.

John says modes would be nice too.

Might be nice to link to this from the countries section of the homepage too.

Just some thoughts.

magicbug, 2019-07-21:
Think it only hides bands with QSOs, but agree needs further functions for filtering by mode, band etc

AndreasK79, 2020-02-21:
I have made something like this in my own project:
[image]
I might be able to incorporate something similar. Let me know, and I'll see what I can do.

dg9vh, 2020-02-21:
I would like to see it 😁

magicbug, 2020-02-21:
@AndreasK79 I think that would be a nice step up improvement to the index.php/awards/dxcc page like @dg9vh would love to see it.

AndreasK79, 2020-03-02:
I have a working version now:
[image]
I still have a few bugs to fix, and some more testing to do.
One but I haven't figured out yet, is the setting of start and end in dxcc_entities. In my table start and end are all set to NULL. I tried to look at the code, and it's supposed to set them, but that does not work at my local copy. Any ideas @magicbug ?

magicbug, 2020-03-02:
Yes entitles has nothing, but prefixes does, think I just copied the table schemas for all three types and just populated what was available

AndreasK79, 2020-03-02:
@magicbug I found the problem. The wrong variable name was used when fetching start and end in update.php. I'll submit the fix with the PR for this. Just need to fix a few more things.

AndreasK79, 2020-03-03:
PR has been submitted. An update for country files needs to be done to set start/end in dxcc_entitites. If this is not done, entitites will not be marked as deleted.

closed 2020-03-04 by magicbug