magicbug/Cloudlog #320: Query QSO count for a given slot via API

M0LTE, 2019-07-21:
**Is your feature request related to a problem? Please describe.**
I'm writing a program which I would like to be able to check against my logbook for needed slots.

**Describe the solution you'd like**
An API equivalent to:
`SELECT COUNT(1) FROM TABLE_HRD_CONTACTS_V01 WHERE COL_DXCC=@dxcc AND COL_BAND = @band;`
I guess something like: `/api/slotqsos?dxcc=230&band=20` returning a bare number.
Mode could be an optional parameter.

**Describe alternatives you've considered**
I am currently querying the database directly but this requires direct access to the DB

**Additional context**
none

magicbug, 2019-07-21:
I'll try to get this added tomorrow afternoon when I've some time between work stuff, so you can carry on with making the lookup Tom.

Do you need this with mode filtering, as I'm guessing you want band= but also mode= ?

If you want a similar thing for grid let me know.

M0LTE, 2019-07-22:
It would be nice to have all of band and mode and grid and for them all to
be optional :-)

Cheers
Tom

magicbug, 2019-07-22:
Hi Tom,

It's now implemented you can call it as 

`/api/country_worked/230/40m/ssb` or `/api/country_worked/230/40m`

It just returns the number not json encoded

magicbug, 2019-07-22:
Same function should work for gridsquare_worked

M0LTE, 2019-07-26:
For "New country on any band" I'm having to do this:

`int qsosWithThatCountryOnAnyBand = 0;
foreach (string metres in new[] { "160", "80", "60", "40", "30", "20", "17", "15", "12", "10", "6", "2", "70c" })
{
    qsosWithThatCountryOnAnyBand += int.Parse(HttpGet($"country_worked/{dxcc}/{metres}m"));
}`

and the database/API is beginning to creak!

Any chance of supporting just `country_worked/dxcc`? :-)

magicbug, 2019-07-26:
@M0LTE How would being able to define band as "all" work?

magicbug, 2019-07-26:
Then you could just use current APIs just tell it you want all the bands rather than a set one

magicbug, 2019-07-26:
@M0LTE You can now do `/api/country_worked/230/all/ssb` and it will check the whole database.

M0LTE, 2019-07-27:
Working, thanks!

closed 2019-07-27 by M0LTE