magicbug/Cloudlog #321: UTF-8 problem

DL3RTL, 2019-07-23:
**Describe the bug**

**Expected behavior**
Name: Jörg

**Frontend**
Name: Jör   (with the missing g to Jörg)

**ADIF export next problem*
<name:4>JÃ¶r

DL3RTL, 2019-07-23:
My ADIF import was originalle <name:4>Jörg

magicbug, 2019-07-23:
Can you try with the latest commit, made sure UTF-8 headers were set correctly.

DL3RTL, 2019-07-23:
I downloaded the version of yesterday.
If I have to install everything again it would take hours :-((
Is there a sprcific file I could exchange?

magicbug, 2019-07-23:
See git commit history, not sure why it would take hours and sadly there's no other way of delivery everything but application/config is safe to overnight, else use Git to do version control.

closed 2019-07-23 by magicbug

DL3RTL, 2019-07-24:
Many thanks for your fast reply.
This morning I downloaded the lastest code and tried again.
Unfortunatley the result is the same.

DL3RTL, 2019-07-24:
How could we open this issue in order to get is debugged? Don't find a button... :-/

DL3RTL, 2019-07-25:
How can we solf this problem ?
Has anyone in idea ?