magicbug/Cloudlog #322: <qsl_sent:1>Y with no effect

DL3RTL, 2019-07-23:
I was uploading an aidf file with the argument
`<qsl_sent:1>Y`

But in the frontend of this specific log entry this outgoing QSL card still has a red arrow. And if I open the entry to edit it the outgoing QSL is 
Sent: No

This seems to be a bug.

magicbug, 2019-07-23:
That is strange, theres no fancy checking going on `<qsl_sent:<?php echo strlen($qso->COL_QSL_SENT); ?>><?php echo $qso->COL_QSL_SENT; ?>` should just grab whats in the database and dump it out.

DL3RTL, 2019-07-23:
As I described: There is a Y. Also in the ADIF export. Just in the frontend there's nothing, so Sent: No.

DL3RTL, 2019-07-23:
ADIF export from Cloudlog:
<call:6>DL6AAO<band:4>160m<mode:3>FT8<freq:4>1.84<qso_date:8>20190709<time_on:6>200200<time_off:6>200200<rst_rcvd:2>59<rst_sent:2>59<qsl_rcvd:0><qsl_sent:1>y<country:27>Federal Republic Of Germany<eor>

Result in Frontend:
[image]

I imported more than 1000 QSOs. Only the very forst QSOs have the right status of QSL shown. All the others don't work.

DL3RTL, 2019-07-24:
I dropped the database and uploaded the same ADIF file once more. Now the status of the QSL is correct.

closed 2019-07-24 by DL3RTL