magicbug/Cloudlog #327: Error message with Clublog upload

DL3RTL, 2019-07-24:
I followed the steps of
https://github.com/magicbug/Cloudlog/wiki/Clublog-Upload

so I opened the url https://www.xxxxx.de/logbuch/index.php/clublog/upload/daniel

In the upload folder I found the file
clublog5d3819d1c341a.adi

But the frontend shows the following error:
couldn't open file "/www/htdocs/xxxxx/xxxxxl.de//uploads/clublog5d3819d1c341a.adi"Error

magicbug, 2019-07-24:
This isn't a cloudlog issue rather a server permissions, make sure /uploads is writeable and readbale.

closed 2019-07-24 by magicbug

dg9vh, 2019-07-24:
What source is the adif file from?

dg9vh, 2019-07-24:
Had this issue with ucxlog, solved by removing first line.

DL3RTL, 2019-07-25:
The ADIF is handmade and there in no additional row in the header.

DL3RTL, 2019-07-25:
Writable and readable? Is it like 777 ?