magicbug/Cloudlog #329: Error when trying to login

ofadam, 2019-07-25:
I get a 404 error when trying to login after a successful installation – any idea what could be causing this?

magicbug, 2019-07-25:
Hard to know without further information
Are you using the latest master from GitHub? Screenshots might be useful

dg9vh, 2019-07-25:
Hi,

what is the content of $config['base_url'] in /application/config/config.php ?

ofadam, 2019-07-26:
https://ofadam.com/dx/

The initial screen will load, but logging in results in a 404 that WordPress then serves.

dg9vh, 2019-07-26:
I'm getting connection refused...

ofadam, 2019-07-26:
Give it a shot now.

dg9vh, 2019-07-26:
So once again my question above:
what is the content of $config['base_url'] in /application/config/config.php ?

AND:

Do you have any .htaccess-file containing some url-rewriting-rules? I see a difference between the URL on my side and on your side:

https://ofadam.com/dx/user/login
versus
https://cloudlog.dg9vh.de/index.php/user/login

I got that index.php in my url (working ufb) but when adding this to your url at the position, it would be removed by the webserver, so there must be some url rewriting active...

dg9vh, 2019-07-26:
Problem could be caused by Wordpress-rewriting-rules because you are hosting cloudlog as subdirectory within your wordpress-installation?

ofadam, 2019-07-26:
That's what I'm thinking – I will try to reposition as a subdomain and see
if it works.

dg9vh, 2019-07-26:
Give it a try and reply :-)

closed 2019-08-15 by magicbug