magicbug/Cloudlog #331: "Pull LoTW data for me" causes errors

M0LTE, 2019-07-26:
**Describe the bug**
"Pull LoTW data for me" causes errors.

**To Reproduce**
Steps to reproduce the behavior:
1. Populate profile with LoTW username and password
2. Admin -> LoTW Export/Import
3. Pull LoTW data for me -> Import
4. Short pause, then a badly rendered page containing the following:

```
A PHP Error was encountered
Severity: Notice
Message: Trying to get property 'COL_LOTW_QSLRDATE' of non-object
Filename: models/Logbook_model.php
Line Number: 789

A PHP Error was encountered
Severity: Notice
Message: Undefined variable: lotw_table_headers
Filename: lotw/analysis.php
Line Number: 4

A PHP Error was encountered
Severity: Notice
Message: Undefined variable: lotw_table
Filename: lotw/analysis.php
Line Number: 5
```

**Expected behavior**
"Pull LoTW data for me" to work

**Screenshots**
Shout if required.

**Desktop (please complete the following information):**
 - Windows 10
 - Latest Chrome

**Additional context**
Running latest master.
Shout if anything else is required.

magicbug, 2019-07-26:
I'm wondering a few things as errors not overly helpful

- Is /uploads folder writeable
- I'm guessing login data is correct
- is there anything in the apache error logs.

magicbug, 2019-07-26:
OK this appears to be the issue

`  function lotw_last_qsl_date() {
      $this->db->select('COL_LOTW_QSLRDATE');
      $this->db->where('COL_LOTW_QSLRDATE IS NOT NULL');
      $this->db->order_by("COL_LOTW_QSLRDATE", "desc");
      $this->db->limit(1);

      $query = $this->db->get($this->config->item('table_name'));
      $row = $query->row();

      return $row->COL_LOTW_QSLRDATE;
    }`

Now the question is whats in the COL_LOTW_QSLRDATE field that its getting upset when its just looking it up

M0LTE, 2019-07-26:
> Now the question is whats in the COL_LOTW_QSLRDATE field that its getting upset when its just looking it up

All NULL.

M0LTE, 2019-07-26:
> * Is /uploads folder writeable
It's mode 777 by the looks of it, so yes.

> * I'm guessing login data is correct
Just pasted it in from the email I got from LoTW :-)

> * is there anything in the apache error logs.
Nothing relevant that I can see in apache's error.log or access.log. I haven't located the PHP logs yet...

M0LTE, 2019-07-26:
Of note - I only just signed up for LoTW and hadn't uploaded any QSOs to my account yet, before hitting the Import button.

I just hit the Import button again, and I got one PHP notice but this time it appeared to work, and I have LoTW confirmations in my log.

magicbug, 2019-07-26:
Thats more like it, its been a while since I looked at the LOTW code so guess i need to check the pull so it errors if theres naff all in the ADIF.

closed 2019-08-15 by magicbug

chris007de, 2021-07-17:
Sorry for reviving this...But FYI, I encounter the same behaviour with a recent version of cloudlog.
I've got a new LoTW account with no contacts filed, so pretty sure it gets an empty ADIF.

When hitting "Manual Sync" it errors with

```
LoTW Matches
A PHP Error was encountered
Severity: Notice

Message: Trying to get property 'COL_LOTW_QSLRDATE' of non-object

Filename: models/Logbook_model.php

Line Number: 1409

LoTW Downloading failed either due to it being down or incorrect logins.
```

Credentials are correct, Upload folder is writeable.

Running the cronjob
```
<myserver>/index.php/lotw/lotw_upload
```
fails with the same error.

Any workaround or do I have to wait until I have a contact in my Lotw profile?

Thanks a lot!